Marus/cortex-debug #104: Example of advanced decoder

diggit, 2018-11-02:
Hi,
could you provide some small example of "advanced" decoder?
Can such decoder send output only to plots or even to console?

Marus, 2019-01-20:
I do have one on my side, will try to package it up and document it on the wiki soon.

diggit, 2019-10-04:
Could you please provide any example. I am not javascript pro and reverse engineering what should decoder class look like is quite difficult for me...

diggit, 2020-04-06:
Hi @Marus @haneefdm, do you have some working advanced decoder or Docs for it? I'd really love to decode some traces.

Marus, 2020-04-06:
I'll try to write up one of my examples that I have later this week.
It uses data transferred in a protocol buffer format.

diggit, 2020-06-16:
@Marus any chance to get some example? :blush:

mcejp, 2020-10-06:
Hi @Marus, that sounds really useful, I'd be very interested in this as well. Anything we can do to help?

abc528, 2021-04-16:
Hope to be answered

haneefdm, 2021-04-16:
This is what I see...

https://github.com/Marus/cortex-debug/wiki/SWO-Output

PhilippHaefele, 2022-01-05:
Hint for (me) writing the documentation for this. The Advanced decoder must implement the interface defined in https://github.com/Marus/cortex-debug/blob/eda3058e580f70af9b512d0c2307a97d8d70ecd6/src/frontend/swo/common.ts#L87 and then it should be enough to set the path to this javascript implementation (file, module) into the `"decoder"` option + setting the `"ports"` option. `"config"` can be used to send a JSON object into init function for custom config properties.

kusstas, 2022-12-29:
Hello, this topic is still actual for me, do you have any template for this?

e-sr, 2023-03-01:
Hi for me is this topic still important, 
I did the following:
- setup the config in lauch.json
- created a decoder .ts file transpiled then in js

premise:
- I first time i try to integrate somethings in  vscde
- beginner in .js .ts...

**launch.json**
```json
...
"rttConfig": {
  "enabled": true,
  "address": "auto",
  "decoders": [
    {
      "port": 0,
      "type": "console",
      "label": "RTT Channel 0"
    },
    {
      "ports":[0],
      "type": "advanced",
      "decoder": "${workspaceFolder}/extern/debug_rtt/advancedecoder/decoder.js",
      "config":{
        "ports":[0],
        "type": "advanced",
        "var1":"this is var 1.",
        "graphid":"id"
      },
    },
  ]
},
...
```
**decoder.ts**
```ts
import * as vscode from 'vscode';
import { SWOAdvancedDecoderConfig } from './common';
import { AdvancedDecoder } from '../common';
import * as fs from 'fs';

class MYAdvancedDecoder implements AdvancedDecoder {
    private output:(output: string)=>void;
    private cc:number;
    private count:number;
    private graphId: string;

    init(
        config: SWOAdvancedDecoderConfig,
        outputData: (output: string) => void,
        graphData: (data: number, id: string) => void):void
        {
            this.output=outputData;
            this.count=0;
            this.cc=10;
            this.graphId=config.graphId;
        }
    
    public softwareEvent(port:number,data: Buffer) {
        this.count++;
        this.output(`${this.count},${data.length}: ${data.buffer}\n`);
    }

    public synchronized() {}
    public lostSynchronization() {}
    public typeName():string{return "advanced";};
    public outputLabel(): string{return "rtt";}
}

export default MYAdvancedDecoder;
```

### remarks 
- As proof of concept it is working. I see the output of  `this.output('${this.count},${data.length}: ${data.buffer}\n');` in the OUTPUT terminal.

**but**

- `this.output` can not be used in `init` method because is uninitialized. I dont'know if it was designed to do so...what is exactly the meaning of using OUTPUT vs. TERMINAL
- at any new start a new OUTPUT is created, the old remain in the list...how can i close the OUTPUT?
- Some Hints where to start using a terminal instead of OUTPUT?

haneefdm, 2023-03-01:
@e-sr Would you mind editing the Wiki and adding a section for this topic. Thnaks btw.

https://github.com/Marus/cortex-debug/wiki/SWO-Output

Question though, you have references to items from `./common`.  Did, you make a copy of our common.ts file? People may want to know how you compiled your ts in js. You can attach a tar.gz file of your sample project if you like.

I have never looked at this area myself, but some comments

> s proof of concept it is working. I see the output of this.output('${this.count},${data.length}: ${data.buffer}\n'); in the OUTPUT terminal.

Not sure what you mean. Do you have a screenshot? What is an OUTPUT terminal. It appears it is going to an OutputChannel (one of the items in the OUTPUT tab) which is not a terminal. This was written in the days when Terminals were not flexible. When I ported almost everything to start using Terminals, I did not touch this part as I did not have an example project.

The idea I think is that you are given callbacks/functions to either `outputData` text information or call `graphData` for graphing or none. It is up to you how you transform the data arriving via the `softwareEvent`. You can save the data to a csv file also for instance or stream to another program. Everytime we get data on this port, an event is generated and you are supposed to use it for your own purposes. I don't think you are required to call the `output` function either.

> this.output can not be used in init method because is uninitialized. I dont'know if it was designed to do so...what is exactly the meaning of using OUTPUT vs. TERMINAL

Depends on the version of the TS compiler. It is getting stricter and stricter. I would have a constructor or declare things like this if you want to avoid a constructor. I believe this was written when Typescript did not allow constructors on interfaces

```TypeScript
    private output: (output: string)=>void | undefined;
    private cc = 0;
    private count = 0 ;
    private graphId = '';
```

And then you might get a warning/error on the following line, you can change it to the following
```
        this?.output(`${this.count},${data.length}: ${data.buffer}\n`);
```

I believe you are supposed to use the `id` you are given. Not sure.

```
            this.graphId = id;
```

haneefdm, 2023-03-01:
I will look into redirecting the OUTPUT to a terminal if you are willing to test it. Yes, opening new windows is also problematic. What I might do is to create the terminal on your first output to that terminal in case you never use it. Once created, it will be cleared for every new use and will just stay there until closed.

e-sr, 2023-03-02:
@haneefdm will try to get it working properly and then if the case i will update the documentation. But i will need some time... 
yes of curse with OUTPUT terminal i mean the output tab.

aq1018, 2023-11-28:
@diggit 

I recently made a [TraceDecoder](https://github.com/aq1018/cortex-debug-trace-decoder) that can parse a variety of mixed data types from one or more channels and display it on real time graph.

You can take a look at the [decoder implementation](https://github.com/aq1018/cortex-debug-trace-decoder/blob/main/src/index.ts) as an example. The repository is also based on TypeScript and complies to JS. This can be a good example for those who wants to setup a more complex project.

Also, I encountered an issue in the base `AdvancedDecoder` that prevents graph data being sent to the actual graph. The fix was trivial. You can find the fix in #961

Hopefully this helps.