Marus/cortex-debug #106: Focus is moved to top of call stack when debugging

0ge, 2018-11-19:
**Steps to reproduce**
1. Add a breakpoint to the program.
2. Start new debug session and start execution
3. When breakpoint is hit, click on some other place in call stack

**Expected results**
When clicking a position in the call stack, the editor is moved there and the symbols are displayed in the Variables section.

**Actual results**
The editor briefly moves focus to where its supposed to, but then immediately moves focus back to the line where the breakpoint was hit.

**Workaround**
None

**Environment**
VS Code 1.29.0 (since 1.29.0)
Cortex-Debug 0.1.21
Configuration:
```
"cwd": "${workspaceRoot}",
            "executable": "<path/to/file>",
            "name": "Debug",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "nRF52832_xxAA",
            "interface": "swd",
            "svdFile": "${workspaceRoot}/vendor/nRF5_SDK_12/svd/nrf52.svd",
            "swoConfig": {
                "source": "probe",
                "enabled": true,
                "swoFrequency": 4000000,
                "cpuFrequency": 0,
                "decoders": [
                    {
                        "port": 0,
                        "label": "SWO Output",
                        "type": "console"
                    }
                ]
```

bigbrett, 2018-11-20:
I can confirm this issue too. Even got a nice GIF of it for y'all

[image]

ronitt, 2018-11-23:
I can also confirm that this is an issue. I don't have a helpful gif though :)

ronitt, 2018-11-23:
Can confirm the issue is brought about by the 1.29 version of VS Code. The problem was remedied after I reverted back to version 1.28.

It can be found here: [https://code.visualstudio.com/updates/v1_28]

sov1178, 2018-11-28:
The same problem here:
[image]

VSCode 1.29.1
Cortex-Debug 0.1.21

The partial workaround (a bit ugly) is to open the window with the breakpoint on the separate group:
[image]

In such UI configuration you can switch to other tabs for viewing/editing.

The problem is a bit annoying...

sov1178, 2018-11-28:
It looks like I have find a fix.  Locate `out/src/frontend/extension.js` file in the extension directory. Comment line 70. The modified code fragment should look like:
```
    activeEditorChanged(editor) {
        if (editor !== undefined && vscode.debug.activeDebugSession && vscode.debug.activeDebugSession.type === 'cortex-debug') {
            const uri = editor.document.uri;
            if (uri.scheme === 'file') {
//                vscode.debug.activeDebugSession.customRequest('set-active-editor', { path: uri.path });
            }
            else if (uri.scheme === 'disassembly') {
                vscode.debug.activeDebugSession.customRequest('set-active-editor', { path: `${uri.scheme}://${uri.authority}${uri.path}` });
            }
        }
    }
```
It seems everything works as it it should work now:
[image]

The corresponding place in extension sources is `activeEditorChanged` handler in the `cortex-debug/src/frontend/extension.ts` file. Comment line 103. It is responsible for this behavior.   

P.S. I am not familiar with the cortex-debug sources, so there might be some side effects after applying this fix, beware of that ;)

Sambo007, 2018-11-28:
@sov1178,

Fantastic!

All we need now is a fix in an official release.

Thanks a lot...

microwavesafe, 2018-11-30:
I can confirm that this fix works. @Marus any chance of adding this into a release?

klaus-liebler, 2018-12-11:
I have the same issue here. I would appreciate to find this fixed soon in a new release. Thank you so much!

paradajz, 2018-12-12:
Another confirmation from my side. @Marus is this extension still being developed?

DBS06, 2019-01-08:
I can also confirm this issue, I will try @sov1178 solution.

Marus, 2019-01-19:
Will be making a release with this fix later today. Sorry for the slow response to these issues - haven't had much time to work on side projects lately.

Marus, 2019-01-20:
I've at least temporarily made the fix that @sov1178 suggested, this was in the 0.2 release that was made yesterday. That may cause some minor issues if you are using the disassembler mode, but will look into a better fix that keeps this issue sorted and solves the problem there.

closed 2019-01-20 by Marus