Marus/cortex-debug #107: cannot cannot read property  getTreeNode of null

bjut-MrZhang, 2018-11-21:
"cannot cannot read property  getTreeNode of null"
breakpoint cannot work

EEVblog, 2018-12-14:
I also have this issue.

chriswils, 2018-12-18:
I get this error as well, but breakpoints are still working. I don't get any updated state of the Cortex peripherals (provided nrf52.svd for an nRF52840 target), so I was thinking it was related.

aibeb, 2019-01-11:
The same issue !

Marus, 2019-01-20:
And yeah - I have to second @rvinci's message - Love your blog @EEVblog - and if you would like to try and sort out the issues you encountered drop feel free to post additional details (like your launch.json and the SVD file being used). I had unfortunately been unable to keep up maintenance on this little side project for a number of months now and am just getting back to it.

stefaandesmet2003, 2019-01-24:
On my system, this error only occurs when I left the 'Cortex Registers' tree list open when disconnecting or stopping the debug session. This view is hidden when no debug session is active (blue status bar), so I close it before ending the debug session. 
Great extension btw!! Haven't figured out all its features yet, but just figured out that 'attach' also starts an openocd process, but doesn't reflash the target each time I restart a debug session. cool!

bigbrett, 2019-03-19:
I too have this error.

haneefdm, 2019-04-08:
I get this message this too. Regardless of JLink or PyOCD or OpenOCD. But not always. I will watch carefully to see what triggers it or how to reproduce it.

haneefdm, 2020-06-20:
Don't thinks this is happening anymore. There is a bit better error tracking as well for SVDs

Please re-open if you wish

closed 2020-06-20 by haneefdm