Marus/cortex-debug #110: `executable` field in launch.json does not support command substitution?..

busticated, 2018-12-11:
hey there :wave:

thanks for a super-helpful extension :pray: i'm building my own extension which pulls in cortext-debug (via `extensionDependencies`) and helps setup and manage `launch.json` on behalf of the user.

to that end, i'm leveraging VSCode's support for commands as configuration settings to make it easier for me to keep `launch.json` in sync with upstream user-driven configuration changes - e.g. 

```json
"svdFile": "${command:extension.getDebuggerSVDFile}",
"armToolchainPath": "${command:extension.getDebuggerCompilerDir}",
```
([vscode docs](https://code.visualstudio.com/docs/editor/variables-reference#_settings-and-command-variables))

all this seems to work fine _except_ in the `executable` field which yields the following error:

```
Invalid executable: /Users/me/path/to/project/${command:extension.getDebuggerELFFile} not found.
```
[[source](https://github.com/Marus/cortex-debug/blob/master/src/frontend/configprovider.ts#L129)]

as you can see, for some unknown reason my command is never called and instead the unsubstituted value is used.

looking at `configprovider.ts`, i don't see anything obvious that would account for the different behavior - `armToolchainPath` works fine for example.

i'm running:
macOS: 10.13.6 (high sierra)
VSCode: 1.29.1
Cortex Debug: 0.1.21

busticated, 2018-12-12:
more info: it seems _none_ of the predefined variables work in the `executable` field ([docs](https://code.visualstudio.com/docs/editor/variables-reference#_predefined-variables-examples)).

which explains this code 👉  https://github.com/Marus/cortex-debug/blob/master/src/frontend/configprovider.ts#L117

but i'm still stumped as to why 🤷‍♂️

busticated, 2018-12-13:
from @weinand over in https://github.com/Microsoft/vscode/issues/64935

> Please note that variable substitution occurs after the resolveDebugConfiguration method of cortex-debug has been run. If their implementation of resolveDebugConfiguration assumes that all values of the launch config already contains resolved values (e.g. paths), then that might explain the. issue you are seeing.

Marus, 2019-01-20:
I'll look into the information that yourself and @weinand passed along regarding the DebugSession.configuration and see if that allows me to resolve this issue.

busticated, 2019-01-20:
thanks! is this fix included in the `v0.2.0` release? i’d be happy to test 👍

Marus, 2019-01-20:
It's in v0.2.1 - was pushed up just a little while ago so should be available soon.

Marus, 2019-01-21:
There was a bug in v0.2.1 - v0.2.2 should have been pushed up now which will hopefully correct the issue introduce in v0.2.1.

Marus, 2019-01-23:
Substitution should be working for executable field in launch.json; I'm going to close the ticket but if you still run into issues please let me know and will investigate further.

closed 2019-01-23 by Marus