Marus/cortex-debug #111: expose `openocdPath` in launch.json?..

busticated, 2018-12-12:
hi there :wave:

very similar request as: https://github.com/Marus/cortex-debug/issues/51 and for the same reasons as mentioned in https://github.com/Marus/cortex-debug/issues/110

would it be possible to expose the `openocdPath` setting in `launch.json` as well?

currently, i end up managing the setting at the workspace level on behalf of my users based on _other_ configuration settings. so the flow is like: 1. tweak seemingly unrelated setting 2. update toolchain paths & dependencies 3. set `cortex-debug.openocdPath` in the `.code-workspace` file based on steps 1 and 2.

instead, being able to do something like this in `launch.json` would be nice:

```
"openocdPath": "${command:extension.getDebuggerOpenOCDPath}",
```

i'm running:
macOS: 10.13.6 (high sierra)
VSCode: 1.29.1
Cortex Debug: 0.1.21

Marus, 2019-01-27:
This should be added to the next release

haneefdm, 2019-06-23:
I am currently using `"serverpath" : "..." ` and it works! If we document and/or add it to IntelliSense, we are golden.

`serverpath` is generic to any gdb-server including openocd since we can only use one server at a time.

haneefdm, 2021-12-22:
For every path mentioned for user/workspace settings, there is now a OS-specific setting. This will work better for users using the sync feature. This is also the method used by VSCode itself. I recommend using this mechanism rather than `serverpath` to keep things portable. From CHANGELOG.md for V0.4.8

> Pathnames for gdb-servers can be OS specific. For instance cortex-debug.openocdPath can be suffixed with one of .linux, .osx or .windows. For instance cortex-debug.openocdPath.windows is used only on Windows and if that is missing, it will default looking for cortex-debug.openocdPath`.

closed 2021-12-22 by haneefdm