Marus/cortex-debug #112: No SVD File Loaded

adammunich, 2019-01-01:
The same "not-loaded" bug seems to appear with this file and the following configuration: https://raw.githubusercontent.com/posborne/cmsis-svd/master/data/STMicro/STM32F405.svd

{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "./build/BLDC_4_ChibiOS.elf",
            "name": "STM32F405 Cortex-Debug (J-Link)",
            "device": "STM32F405RG",
            "interface": "jtag",
            "svdFile": "./debugging/STM32F405.svd", 
            "showDevDebugOutput": true
        }
    ]
}

adammunich, 2019-01-01:
I discovered the problem, turns out, the svdFile expects an absolute directory, unlike the executable...

    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "./build/ChibiOS.elf",
            "name": "STM32F405 Cortex-Debug (J-Link)",
            "device": "STM32F405RG",
            "interface": "jtag",
            "svdFile": "${workspaceFolder}/debugging/STM32F405.svd", 
            "showDevDebugOutput": true
        },

Marus, 2019-01-19:
Good catch - will look into making the svdFile reference work for relative paths as well.

Marus, 2019-01-20:
Next release (0.2.1) should interpret relative svdFile paths properly - using full paths or ${workspaceFolder} should also still work.

closed 2019-01-20 by Marus