Marus/cortex-debug #113: Enhancement: interaction with build tool

RolfNoot, 2019-01-04:
It happens quite often that I start debugging without having the project build.

Is it possible to interact with the build tool (currently CMake Tools / vector-of-bool) to start a build action and wait until it's finished before starting the debugger?

Thanks!
Rolf

perlindgren, 2019-01-04:
You have "preLauchTask". That runs a build task before launching the debugger. I use it together with Rust/Cargo, and it works well.

There is a (minor) issue. If you press restart, the "preLaunchTask" is indeed executed again, BUT, the new binary is not loaded. To that end it would perhaps make more sense to have it either way.

1) Don't re-run "preLaunchTask" on restart. (Just restart the binary as is..., fast and swift restart)
2) Run "preLaunchTask" and load the new binary. (Allows for quick fixing the code without having to close/reopen the debug session.)

Either would be fine, but the behaviour as of now is somewhat inconsistent. (you make changes to the code, it re-compiles, but it runs the old code..) A possible drawback of the second approach is what should happen IF your edited program fails to compile? If you close the session and restarts it, you will have a question, if you want to continue or not. If opting for the second approach, you are kind of in the middle of a debug session already, what would the desired behaviour be? I'm not sure...

Best regards,
  Per

RolfNoot, 2019-01-04:
Thanks Per for your prompt reply.

Thanks I'll look into using the "preLaunchTask".

If it's possible to integrate the two tools together this would be even nicer (llok at my mockup)!

Thanks!
Rolf
[image]

perlindgren, 2019-01-04:
You mean the project outline and the debug view? I'm not sure, `vscode` puts some restrictions on panel layouts AFAIK. Maybe someone with more `vscode` experience has an answer.
/Per

RolfNoot, 2019-01-04:
Thanks Per. One question though. How did you use the "preLaunchTask" to do the rebuild action. Did you call cmake directly?

perlindgren, 2019-01-04:
Hi

You refer to a (build) task. There is more info:

https://code.visualstudio.com/docs/editor/debugging

/Per

RolfNoot, 2019-01-11:
Thanks Per.

Got it working right now.

I agree it would be better if the preLaunchTask is not executed when pressing Restart. 

It would make sense if the preLaunchTask would contain a boolean setting "launchOnRestart: false".

Thanks!
Rolf

Marus, 2019-01-20:
I'll look into this. preLaunchTask is something provided by VSCode - not something that the extension has specifically provided - but may be able to provide some settings that tweak it's behavior (either so we can at least reload the executable, or prevent it from re-compiling).

haneefdm, 2020-06-20:
I don't think this is a problem with this extension. If I am wrong, please re-open or submit a new issue with more details.

closed 2020-06-20 by haneefdm