Marus/cortex-debug #116: EACCES error

rerickson1, 2019-01-10:
Using Cortex-Debug v0.1.21

I received the following error and cannot get the debugger to start anymore.
Unhandled rejection Error: listen EACCES 0.0.0.0:50000
    at Object._errnoException (util.js:1024:11)
    at _exceptionWithHostPort (util.js:1046:20)
    at Server.setupListenHandle [as _listen2] (net.js:1334:19)
    at listenInCluster (net.js:1392:12)
    at doListen (net.js:1501:7)
    at _combinedTickCallback (internal/process/next_tick.js:141:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)

I have gotten around this issue previously by uninstalling the extension, rebooting my PC, and the reinstalling it.  This no longer works for me and I cannot get it to run.  I have not changed any of my launch or config settings and those settings worked for me previously.
Any help debugging this issue would be helpful.

blckmn, 2019-04-03:
I have exactly the same problem, although I suspect it was caused by me enabling VMs and Docker on my machine, and thus I now have a number of Virtual Network Interfaces present.

I think we could get around it by introducing specific IP and PORT parameters so that we can always start up on a specific ip and port.

blckmn, 2019-04-04:
Doing a little more digging on this and the error is in the portastic module.

```Unhandled rejection Error: listen EACCES 0.0.0.0:50000
    at Server.setupListenHandle [as _listen2] (net.js:1269:19)
    at listenInCluster (net.js:1334:12)
    at Server.listen (net.js:1421:7)
    at Object.<anonymous> (D:\source\projects\node_test\node_modules\portastic\lib\portastic.js:46:12)
    at Object.tryCatcher (D:\source\projects\node_test\node_modules\bluebird\js\main\util.js:26:23)
    at Object.test (D:\source\projects\node_test\node_modules\bluebird\js\main\method.js:15:34)
    at D:\source\projects\node_test\node_modules\portastic\lib\portastic.js:123:21
    at tryCatcher (D:\source\projects\node_test\node_modules\bluebird\js\main\util.js:26:23)
    at ReductionPromiseArray._promiseFulfilled (D:\source\projects\node_test\node_modules\bluebird\js\main\reduce.js:105:38)
    at ReductionPromiseArray.init (D:\source\projects\node_test\node_modules\bluebird\js\main\promise_array.js:92:18)
    at ReductionPromiseArray.init (D:\source\projects\node_test\node_modules\bluebird\js\main\reduce.js:42:10)
    at Async._drainQueue (D:\source\projects\node_test\node_modules\bluebird\js\main\async.js:128:12)
    at Async._drainQueues (D:\source\projects\node_test\node_modules\bluebird\js\main\async.js:133:10)
    at Immediate.Async.drainQueues (D:\source\projects\node_test\node_modules\bluebird\js\main\async.js:15:14)
    at runCallback (timers.js:705:18)
    at tryOnImmediate (timers.js:676:5)```

blckmn, 2019-04-04:
Can get the same error with a simple node app: 

```
var portastic = require("portastic");

portastic.find({
    min: 50000,
    max: 52000
}).then(function(ports){
    console.log('Ports available are: %s', ports.join(', '));
}, (err) => {
    console.log(`Failed to find open ports: ${err.toString()}`);
});
```

blckmn, 2019-04-04:
https://github.com/Marus/cortex-debug/pull/146

To provide a work around.

rerickson1, 2019-04-18:
How can I get a test version of the plugin to test this out?

haneefdm, 2019-05-28:
My latest PR #169 may fix this problem. It replaces `portastic` with another method. It would be nice to see if this fixes it. If not, I can work on it.

If you want to try it out you can get it at https://github.com/haneefdm/cortex-debug

rerickson1, 2019-05-28:
I am testing your changes now. So far I see no issues.  I will try tonight at home to see if the EACCESS is solved.  My PC only exhibits the issue when I am not on the corporate network.

haneefdm, 2019-05-28:
@rerickson1 I made a couple of changes to log when we get EACCES error. We should only see it if you don't have permission to use that port (as opposed to EADDRINUSE if someone else is using that port). In both cases, I consider that the port is not available for use. You may want to fetch again to get that change.

Marus, 2019-06-09:
@haneefdm's updates to replace portastic should be available in the new v0.2.7 release - please try that out and see if this fixes your issue with the port testing.

rerickson1, 2019-06-24:
Great work to all.  I can confirm the issue is solved.  At this time I am running v0.3.1 and this issue no longer occurs.

closed 2019-06-24 by rerickson1