Marus/cortex-debug #117: GDB port conflict Again

slademcthornbody, 2019-01-18:
Having same issue as per #24 on V0.1.21. 

Found many other apps were claiming port 50000 and cortex-debug was still trying to open on port 50000 and failing.

I had to close about 3 apps - when I closed each one, another would claim 50000 (itunes, dropbox etc)

I think a setting to override the port to a fixed value (JLINK defaults to 2331) would help.

Thanks

Marus, 2019-01-20:
Will try to look into this - it should be checking to see if the ports are free before trying to use them - so it's a bit strange. The reason I don't use a hard-coded one (or the particular probes defaults) is that it causes issues if you have two simultaneous debug sessions at the same time.

What OS are you working on - it could be that the library that I'm using to check to see if the ports are in use is not working properly on that OS.

paradajz, 2019-01-20:
Just to chime in - I had this as well on latest Windows 10 update (october one?) and latest version of VScode as well.

Marus, 2019-01-20:
It's possible it is something with the `portastic` library (which I used for determining if ports are in use) is not always working on Windows (I do my development primarily on macOS and sometimes Linux). Out of curiosity, have you tried running VSCode as administrator? I'm just wondering if it may be a permissions issue of some sort (maybe because what is using the ports already is running as a different, perhaps Windows system, user).

paradajz, 2019-01-20:
I'll let you know tomorrow - I only use Windows at work.

slademcthornbody, 2019-01-20:
Thanks. I am using windows 10 64-bit. I tried running as administrator but had the same issue.

Marus, 2019-01-20:
Ok - I’ll try getting setup in a win 10 environment and see what I can
figure out. What things other than iTunes seemed to cause conflicts, just
trying to make sure I have things that may recreate the problem.

slademcthornbody, 2019-01-21:
Thanks - also had Dropbox and lifesize video conf software claim the port.

Marus, 2019-01-21:
Thanks for the info. Will try to get a Win 10 VM setup to try and determine why it is not properly detecting that the ports are in use and selecting a different one; may also add a workspace setting as well to override the starting port that is searches from; currently it should select the first 1 to 3 (depending on the particular options and how many ports it needs) available ports starting at 50000, but apparently in some cases it's missing that key "available" word.

paradajz, 2019-01-21:
No difference with Admin run as well.

nergnezor, 2019-02-06:
I have the same issue since updating to Windows 18.09.
I checked used ports and couldn't see anything using 50000, but I might have missed something

slademcthornbody, 2019-02-11:
Any progress on this issue? Working on getting the environment setup so I can build this extension and modify

slademcthornbody, 2019-02-15:
Have added an optional port range for portastic that works around this issue. Pull request added.

noppej, 2019-03-07:
Before using VSCode and your amazing extension, I used to use SW4STM. They allowed a config parameter to set the gdb port number. The primary benefit is that this allowed me (and others) to debug multiple boards simultaneously, by having multiple OpenOCD processes running - one for each board. Will your port cycling method allow this? (I tried it with the current Insiders build, and it gives errors on the second board because port 5000 is already in use).

noppej, 2019-05-13:
Hi @Marus,

In the comment referenced below, you suggest that you can have two simultaneous debug sessions.  I have two boards, a STM32F4 and a STM32F7. If I try to start a debug session on either board while the other is already in a debug session, I get an error ...
```text
Info : STM32F746NGHx.cpu: hardware has 8 breakpoints, 4 watchpoints
Error: couldn't bind gdb to socket on port 50000: Address already in use
shutdown command invoked
```
Any pointers/suggestions?
PS. Using v 0.2.4 of your plugin

> Will try to look into this - it should be checking to see if the ports are free before trying to use them - so it's a bit strange. The reason I don't use a hard-coded one (or the particular probes [...]

Marus, 2019-06-09:
The port selection issues should hopefully be fixed in the new 0.2.7 release which has big improvements for port selection from @haneefdm

closed 2019-06-09 by Marus

fellerts, 2019-06-27:
Hi @Marus,

I began to experience the same issue on Ubuntu 16.04 after upgrading to version 0.2.7, where I get a popup saying `Failed to find open ports: Error: listen EADDRNOTAVAIL ::1:50000.` Running `sudo netstat -tulpn` does not indicate port 50000 being used by another process, and I have no other instances of ``JLinkGDBServerExe running. Running VSCode as root does not help. I don't remember seeing this problem before 0.2.7.

Downgrading to 0.2.6 seems to resolve the problem. I'm happy to supply additional information if that helps!

haneefdm, 2019-06-27:
@Marus, @fellerts PR #176 may fix this issue. The failure can happen on systems that do not have any kind of IPv6 support at all. In this PR we only check to see if IPv4 ports are free and that too only loopback ones. It also ignores VPN tunnels

marduino, 2019-12-10:
Hi, sorry for jump in to this old topic, may I ask a silly question that is there any plan to add a configuration to allow my launch to set these ports' number?

Why I am asking is because I actually have 3 different apps, one of them can't be programed to use 5000 because of certain reason, allow my customization on the port could significantly reduce my complexities.

haneefdm, 2019-12-10:
I don't know if there are plans for users to provide port numbers. But, if port 50000 is in use, cortex-debug should be avoiding it. Maybe I can help with that. May I ask if you are seeing a problem with this?

slademcthornbody, 2019-12-10:
> Hi, sorry for jump in to this old topic, may I ask a silly question that is there any plan to add a configuration to allow my launch to set these ports' number?
>
> Why I am asking is because I [...]

You can use my fork - I made a mod to allow port number to be set

marduino, 2019-12-11:
> You can use my fork - I made a mod to allow port number to be set

Thanks Slade, I will check it out, btw, I found this gnu debugger "https://marketplace.visualstudio.com/items?itemName=metalcode-eu.gnu-debugger"
It is just simply allow me to set the clientArgs and serverArgs passing to the 2 program. I think it is very nice. Would you also add something like that?

haneefdm, 2022-02-07:
What about TCP ports for SWO, RTT, a 2nd or 3rd core, multiple boards, etc. There is a lot more to it. 

The gdbTarget is for people who... well you already know what that is.

Sorry, this is not up for debate. This is how we do it.