Marus/cortex-debug #119: 0.2.x can NOT work

long568, 2019-01-21:
macOS Mojave 10.14.2
VScode(1.30.2) + OpenOCD(0.10.0) + ST-LINKV2
launch.json
`{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/HiMadOS.elf",
            "name": "OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "interface": "swd",
            "preLaunchTask": "build",
            "configFiles": [
                "interface/stlink-v2.cfg",
                "target/stm32f1x.cfg"
            ],
            "runToMain": true
        }
    ]
}`
Cortex-Debug_0.1.21 works fine.
Cortex-Debug_0.2.x can not launch OpenOCD and alway get  launch timeout error. 
When using Cortex-Debug_0.2.x, the ST-LINK has no action at all.
I'm developing a RTOS(https://github.com/long568/MadOS), you can check the Settings.

EvilEli, 2019-01-21:
[I think the issue is located here](https://github.com/Marus/cortex-debug/commit/99018d7f5e644d1f11eea2aface4d8073cac7e21?_pjax=%23js-repo-pjax-container#r32006865)

Marus, 2019-01-21:
Should be a v0.2.3 shortly that will correct the OpenOCD issue. Thanks for the reference @EvilEli

long568, 2019-01-21:
Thanks a lot.
So I will use the 0.1.21 until 0.2.3 released.

Marus, 2019-01-21:
0.2.3 should be available now if you look for an update.

Marus, 2019-01-21:
I'm going to assume that 0.2.3 fixed your issue - if it is still not working please reopen the ticket.

closed 2019-01-21 by Marus

long568, 2019-01-21:
I just tested 0.2.3, and it's working fine now.
Thanks a lot.