Marus/cortex-debug #125: connect to a running openocd session instead of launch

eco747, 2019-02-08:
as the title said, is there any change to connect to a running openocd session instead of launching a new one ?
by the way having a way to specify a port should be great

Marus, 2019-02-10:
Currently there isn't a way to do this - something I'm considering for the future. The issue is that with an external OpenOCD session the debugger interface cannot easily know for sure what the settings and current state of the debug session it is attaching to will be - so can get things in the UI out of sync with the actual GDB state.

If you are trying to debug something that is already running on a board you can use the `attach` launch type which while it will still launch OpenOCD itself, will not reset the target processor nor flash memory (you will need to ensure that your binary used for debugging matches what is actually flashed on the target) - but simply halts the target processor when it first connects.

haneefdm, 2019-06-23:
I think this is fixed with version 0.2.7 when the 'serverType`: "external" and the "gdbTarget" setting specify host and port? At least one user has reported this as working. See #66

I would close this issue. Perhaps documentation is needed but it is mentioned in the ChangeLog.

superjax, 2019-10-30:
Just as a word of confirmation,  I couldn't get the normal way to work, so I trie dthis, and it seems a lot more reliable.  Here is my `launch.json`

closed 2020-06-20 by haneefdm