Marus/cortex-debug #128: tpiu external when swo source is a serial port

diegoherranz, 2019-02-16:
If you are using an external UART port for getting the SWO data and you use a configuration similar to this on `launch.js`:

```
"swoConfig":{
		"enabled": true,
		"source": "/dev/ttyACM1",
		"swoFrequency": 1000000,
		"cpuFrequency": 8000000
                }
```

I would expect the `tpiu` command to openocd to say `external` but the only place in the code where I can find a `tpiu` command is the following and it's fixed as `internal`:

https://github.com/Marus/cortex-debug/blob/5fe9ac69f84a9ef6e88352137423ed69eb9a6c81/src/openocd.ts#L145

Shouldn't it be `external` if a serial port is defined on `source` and `internal` if `"source": "probe"` is used?
Am I missing something?

Many thanks!

diegoherranz, 2019-02-16:
I've modified `openocd.ts` like the following and it works for me with an external UART adapter.

```
let tpiuIntExt;
if (this.args.swoConfig.source === 'probe') {
    tpiuIntExt = `internal ${this.swoPath}`
}
else {
    tpiuIntExt = `external`
}
commands.push(`tpiu config ${tpiuIntExt} uart off ${this.args.swoConfig.cpuFrequency} ${this.args.swoConfig.swoFrequency}`);
```

Does it sound OK to you? Would you accept a patch/PR? If so, please let me know about the variable name preference and or javascript specifics (not a javascript expert by any means).

Thanks!

Marus, 2019-05-15:
Hi - sorry for the slow response - hadn't been working on this project much recently. I do believe your general idea there is correct. I'll look at including it in an upcoming release.

diegoherranz, 2019-05-15:
No problem. And thanks for this tool!

diegoherranz, 2019-05-18:
Fixed on https://github.com/Marus/cortex-debug/pull/164

closed 2019-05-18 by diegoherranz