Marus/cortex-debug #130: TypeError: Cannot read property 'scope' of null when stepping over

naigonakoii, 2019-02-22:
When debugging with the newest version of Visual Studio Code (1.31) using Cortex-M with OpenOCD, every time I click step over I get the following exception:

Could not step over: TypeError: Cannot read property 'scope' of null

I've attached the callstack. This happens anywhere in the file. Breakpoints do hit okay and the debugger seems to be attached correctly, just step doesn't work which makes debugging pretty difficult.
[-1550812122361.log](https://github.com/Marus/cortex-debug/files/2892477/-1550812122361.log)

Flojolomo, 2019-02-28:
Same problem here. 
@naigonakoii a workaround is to startup openocd through a terminal and to debug with a GDB configuration. Here is the link how to do it: http://www.justinmklam.com/posts/2017/10/vscode-debugger-setup/
Unfortunately you have to build and flash every time before you run the configuration. That's the big advantage of the cortex-debug environment

Marus, 2019-05-16:
@naigonakoii - Hi if you're still trying to get this to work can you provide me more information about your setup - in particular can you provide your launch.json and what build/where you've sourced your OpenOCD version (there are a lot of customized versions of OpenOCD floating around).

Also note that V0.2.4 changed some requirements for OpenOCD - so you may need a more recent build. We recommend building from source yourself (`brew install open-ocd --HEAD` will work for an easy install on macOS with homebrew) or using the releases from gnu-arm-eclipse

haneefdm, 2020-06-20:
No activity/response, closing issue. Please re-open if you wish

closed 2020-06-20 by haneefdm