Marus/cortex-debug #134: Connected to target but nothing happens

paradajz, 2019-02-27:
Here's my launch config:

```
{
        "runToMain": true,
        "type": "cortex-debug",
        "request": "launch",
        "servertype": "jlink",
        "cwd": "${workspaceRoot}",
        "executable": "./build/TestApp.elf",
        "name": "Debug (J-Link)",
        "device": "EFM32JG1B100F256GM32",
        "interface": "swd",
        "ipAddress": null,
        "serialNumber": null,
        "svdFile": "${workspaceFolder}/misc/EFM32JG1B100F256GM32.svd",
        "showDevDebugOutput": true
}
```

When I start the debugging, this is all I get in adapter output:

```
SEGGER J-Link GDB Server V6.40 Command Line Version
JLinkARM.dll V6.40 (DLL compiled Oct 26 2018 15:08:28)

Command line: -if swd -port 50006 -swoport 50007 -telnetport 50008 -device EFM32JG1B100F256GM32
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50006
SWO raw output listening port: 50007
Terminal I/O port:             50008
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 EFM32JG1B100F256GM32
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: Energy Micro EFM32 compiled Apr 20 2016 12:17:23
Hardware: V1.00
S/N: 440068203
Feature(s): RDI
Checking target voltage...
Target voltage: 3.60 V
Listening on TCP/IP port 50006
Connecting to target...Connected to target
```

After that nothing happens - debugger doesn't stop on any breakpoint. It's as if the process launches and then stops.

VS Code info:

```
Version: 1.31.1
Commit: 1b8e8302e405050205e69b59abb3559592bb9e60
Date: 2019-02-12T02:19:29.629Z
Electron: 3.1.2
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Linux x64 4.15.0-45-generic
```

OS is Ubuntu 18.04.

Output from debug console:

```
1-gdb-set target-async on
2-interpreter-exec console "source /home/paradajz/.vscode/extensions/marus25.cortex-debug-0.2.3/support/gdbsupport.init"
3-target-select extended-remote localhost:50006
4-interpreter-exec console "monitor halt"
5-interpreter-exec console "monitor reset"
6-target-download
7-interpreter-exec console "monitor reset"
8-enable-pretty-printing
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from /home/paradajz/dev/git/TestApp/build/TestApp.elf...\n"}]}
Reading symbols from /home/paradajz/dev/git/TestApp/build/TestApp.elf...
-gdb-exit
```

paradajz, 2019-02-27:
Hm, seems to work fine after I get rid of link-time optimization flags in my build process.

haneefdm, 2019-05-15:
@paradajz Hmmm, I see the command '-gdb-exit'. That is only sent by Cortex-Debug when hit the red-button to stop debugging. Sounds like gdb is exiting by itself because it does not like something in the elf file.

Do you mind starting gdb from the command line and see what happens.
```
gdb -q --interpreter=mi2 /home/paradajz/dev/git/TestApp/build/TestApp.elf
```
Make sure you are using the right gdb though

haneefdm, 2020-06-20:
No response, closing issue. Please re-open if you wish

closed 2020-06-20 by haneefdm