Marus/cortex-debug #137: How to pass parameters to GDB?

overheat, 2019-03-12:
Hi,

I am wondering that if possible to pass "set path-substitution xxx xxx" parameter to gdb?

BTW, thanks for your Cortex-debug project 👍

clementperon, 2019-03-19:
Also interested in this feature !

And also thank you very much for the project :)

bigbrett, 2019-03-19:
This is pretty basic functionality that most GDB users require. I would like to please see this supported.

haneefdm, 2019-04-08:
Yes, [source maps](https://sourceware.org/gdb/onlinedocs/gdb/Source-Path.html) are important.

In general, we want to pass additional args to gdb.

In this case though, can't we do this with a [.gdbinit](https://sourceware.org/gdb/onlinedocs/gdb/gdbinit-man.html) file either in the current dir or in your home dir?

haneefdm, 2019-04-12:
The option "debuggerArgs" already exists and you can put that in your launch.json. Kinda like

`            "debuggerArgs": [
                "-abcd",
                "optionarg"
            ]
`

or whatever makes gdb happy without disturbing the debug session. If this works for you, perhaps you can close the issue.

Marus, 2019-06-09:
I think there are already available to do this:

1) as @haneefdm mentioned there is the `debuggerArgs` option in launch.json - these will all be passed to GDB on starting.
2) If you want to send commands to GDB as part of launch you can use the `preLaunchCommands` and `postLaunchCommands` (there are variants for attach and restart as well; also override versions to replace the default commands that are run).
3) You should be able to use a .gdbinit file in your current dir or home dir (may depend on options your GDB was compiled with)

closed 2019-06-09 by Marus