Marus/cortex-debug #139: #FeatureRequest

noppej, 2019-03-14:
Hi @Marus 
Do you have any documentation and/or examples for configuring graphing output?  I'd be happy to take anything rough and help clean it up and resubmit if that will help.

Keriz, 2019-04-30:
Hello @Marus ,
I've been trying to setup the graphing output but couldn't get it to work so far. Reading package.json I tried to setup a preliminary documentation.

You can find it there: https://wiki.octanis.org/stm32/vscode#using_the_swo_pin_debugging_with_graphics

So I have no window that opens up in VS Code to display a graph. I went and opened the following file in my browser:
`file:///C:/Users/guillaume.thivolet/.vscode/extensions/marus25.cortex-debug-0.2.3/grapher/index.html?port=53333`

Starting the debugging, I get something in the browser:
[image]

But it doesnt look like data is being sent. 
Here is what I used to write something on the ITM.

`ITM->PORT[1U].u32 = (uint32_t) value;`

Could you please provide working examples, code to read, and a few lines on how to get the graphics running? Thanks!

noppej, 2019-05-03:
Hi Keriz,
With the aid of your documentation, and some other tweaks, I was able to get this to work, using the same write to ITM, I directed my board's CPU% to the chart as a means to demonstrate. 

Firstly, to open the output in VSCode, you have to replace some of the extension code in "/Users/whatever/.vscode/extensions/marus25.cortex-debug-0.2.3/out/src/frontend/swo/core.js". The extension used the deprecated 'vscode.previewHTML', which no longer works. I just dropped the following in place of the line that has the previewHTML line ...
```Javascript
                //To do this, please add a 'require' for path at top of file
                const grapherFilePath = vscode.Uri.file(
                    path.join(extensionPath, 'grapher', 'index.html')
                );
                const grapherResourcePath = grapherFilePath.with({ scheme: 'vscode-resource' });
                const panel = vscode.window.createWebviewPanel(
                    'SWO Graphs',
                    `SWO Graphs [${time}]`,
                    vscode.ViewColumn.One,
                    {
                        preserveFocus:true,
                        enableScripts:true
                    }
                );
               panel.webview.html = `
                <!DOCTYPE html>
                <html lang="en">
                <head>
                    <meta charset="UTF-8">
                </head>
                <body style = "height: 100vh;">
                    <object type="text/html" data="${grapherResourcePath}?port=${port}&timestamp=${timestamp}" height="100%" width="100%">
                    </object>
                </body>
                </html>`;
```

Once you've done this, follow your instructions, with some minor differences .... 

- the default graph line is white on white on white. Fix it in launch.json with something like 
```json
 "graphConfig": [
                {
                    "label": "CPU Monitor",
                    "type": "realtime",
                    "maximum": 100,
                    "minimum": 0,
                    "timespan": 180,
                    "plots": [
                        {
                            "label": "CPU %",
                            "graphId": "CPU_Data",
                            "color": "#8A2BE2"
                        }
                    ]
                }
            ]
```
- I couldn't get the data to start feeding, until I added the "type":"realtime" shown above
- Finally, I had problems with ITM registry settings getting messed up and intermittent behaviour, so I added the following lines to my openocd ".cfg" file, to force the itm ports I need to open ...
```text
$_TARGETNAME configure -event gdb-detach { itm port 0 off; itm port 1 off; reset; shutdown }
$_TARGETNAME configure -event reset-start {adapter_khz 4000; itm port 0 on; itm port 1 on }
```

The result shows up as a tab in VSCode, with my output shown in this screenshot ...
[image]

If you still can't get it to work, let me know and we can compare configs until we find the difference.

goosenphil, 2019-05-04:
> 
> 
> Hi Keriz,
> With the aid of your documentation, and some other tweaks, I was able to get this to work, using the same write to ITM, I directed my board's CPU% to the chart as a means to demonstrate.
> 
> Firstly, to open the output in VSCode, you have to replace some of the extension code in "/Users/whatever/.vscode/extensions/marus25.cortex-debug-0.2.3/out/src/frontend/swo/core.js". The extension used the deprecated 'vscode.previewHTML', which no longer works. I just dropped the following in place of the line that has the previewHTML line ...
> 
> ```js
>                 //To do this, please add a 'require' for path at top of file
>                 const grapherFilePath = vscode.Uri.file(
>                     path.join(extensionPath, 'grapher', 'index.html')
>                 );
>                 const grapherResourcePath = grapherFilePath.with({ scheme: 'vscode-resource' });
>                 const panel = vscode.window.createWebviewPanel(
>                     'SWO Graphs',
>                     `SWO Graphs [${time}]`,
>                     vscode.ViewColumn.One,
>                     {
>                         preserveFocus:true,
>                         enableScripts:true
>                     }
>                 );
>                panel.webview.html = `
>                 <!DOCTYPE html>
>                 <html lang="en">
>                 <head>
>                     <meta charset="UTF-8">
>                 </head>
>                 <body style = "height: 100vh;">
>                     <object type="text/html" data="${grapherResourcePath}?port=${port}&timestamp=${timestamp}" height="100%" width="100%">
>                     </object>
>                 </body>
>                 </html>`;
> ```
> 
> Once you've done this, follow your instructions, with some minor differences ....
> 
>     * the default graph line is white on white on white. Fix it in launch.json with something like
> 
> 
> ```json
>  "graphConfig": [
>                 {
>                     "label": "CPU Monitor",
>                     "type": "realtime",
>                     "maximum": 100,
>                     "minimum": 0,
>                     "timespan": 180,
>                     "plots": [
>                         {
>                             "label": "CPU %",
>                             "graphId": "CPU_Data",
>                             "color": "#8A2BE2"
>                         }
>                     ]
>                 }
>             ]
> ```
> 
>     * I couldn't get the data to start feeding, until I added the "type":"realtime" shown above
> 
>     * Finally, I had problems with ITM registry settings getting messed up and intermittent behaviour, so I added the following lines to my openocd ".cfg" file, to force the itm ports I need to open ...
> 
> 
> ```
> $_TARGETNAME configure -event gdb-detach { itm port 0 off; itm port 1 off; reset; shutdown }
> $_TARGETNAME configure -event reset-start {adapter_khz 4000; itm port 0 on; itm port 1 on }
> ```
> 
> The result shows up as a tab in VSCode, with my output shown in this screenshot ...
> [image]
> 
> If you still can't get it to work, let me know and we can compare configs until we find the difference.

Awesome! I think this should be added to the documentation.

Marus, 2019-06-09:
Thanks - I'm trying to get to some documentation and start adding things to the Wiki section; if either of you would like to contribute your knowledge on the topics to the wiki pages that would be greatly appreciated.

ppryga, 2020-06-02:
Hi guys! 

You have provided great tool! Congratulations!
I'm new user, nevertheless I want the project to progress and be better and more comfortable. 

I was wondering about debugger user experience. You probably know how great is Visual Studio debugger in day-to-day usage. VS code debugger for native application is also great. 
In Cortex Debugger I miss a few things:
- variables vales are shown in tooltips but it is limited. Especially in case of pointers, usually I see address where is points to. I would be more useful if tooltip has possibility to show what is under cover of pointer, e.g. expand structures, show array if  it points to, expand string value. 
- it would be also good to have a drag-and-drop of variables to watch window.

These are first two things that came to my attention recently. 

Greetings,
Piotr

PhilippHaefele, 2022-08-28:
@ppryga-nordic Can you please create two new issues for your feature requests, as they are somehow unrelated here.
This will help us sorting things as we can close this issue and discuss the two features in separate threads.
Thanks 👍

haneefdm, 2022-08-29:
> it would be also good to have a drag-and-drop of variables to watch window.

We don't own these panels and in general the UI, let alone an interaction between two different panels. VSCode does. You have to submit an issue with VSCode. But, I don't think right clicking and selecting 'Add to Watch' is even faster.

I do not like attaching random thoughts to other un-related issues.

Please submit a new issue for the tooltips and describe which tooltip. This may also be another UI thing we can't do

I am closing this as there is not much else to do here. New issues are welcome