Marus/cortex-debug #140: vscode problem 1.32.3

perlindgren, 2019-03-20:
Hi Marus

When debugging a Rust application on stm32f103 (bluepill), I now get the following errors.

Cannot read property 'getTreeNode' of null

Error: e is not iterable
Error: Cannot read property 'Symbol(Symbol.iterator)' of undefined
Error: Cannot read property 'Symbol(Symbol.iterator)' of undefined
Error: Cannot read property 'Symbol(Symbol.iterator)' of undefined

Vscode was recently updated to :'
code -version
1.32.3
a3db5be9b5c6ba46bb7555ec5d60178ecc2eaae4
x64

(as packaged by arch linux).

My build task:
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558 
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "type": "shell",
            "label": "cargo build --example hello --release",
            "command": "cargo build --example hello --release",
            "problemMatcher": [
                "$rustc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
    ]
}

and launch:

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "name": "hello",
            "preLaunchTask": "cargo build --example hello --release",
            "executable": "./target/thumbv7m-none-eabi/release/examples/hello",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f1x.cfg"
            ],
            "postLaunchCommands": [
                "monitor arm semihosting enable"
            ],
            "cwd": "${workspaceRoot}"
        },
    ]
}

It kind of works mut the Notifications keep popping up, and I have experienced some sporadic crashes, but its hard to replicate, so not sure the reason.

Best regards,
  Per

Marus, 2019-05-18:
What version of the cortex-debug extension are you using?

timboldt, 2019-05-19:
Try compiling in debug mode (without `--release`).

ninjasource, 2019-06-02:
Hi Tim,

Compiling without --release works (the debug symbols are loaded correctly) BUT that's not an option for large code bases. The stm32f103 is a very constrained device (64KB flash) and, at some point, you have to compile --release to fit the binary on the device. In the ./cargo/config file I set the following:

```
[profile.release]
debug = true
```

but this still goes straight to assembler when you start a debug session. Any ideas?
David

timboldt, 2019-06-02:
My solution was to compile in debug mode with some optimization, e.g.

```
[profile.dev]
opt-level = 1
```

`opt-level=1` doesn't seem to mess things up too much. I've seen other people use `opt-level = s` as well.

ninjasource, 2019-06-02:
Thanks, this worked for me in debug mode (opt-level=1 was too large):

```
[profile.dev]
opt-level = 's'
```

closed 2020-06-20 by haneefdm