Marus/cortex-debug #142: Question: Help needed for debugging the gdb/gdb-server part of the extension

haneefdm, 2019-03-26:
I am trying to debug the Cortex-Debug Extension itself but really, just the GDB (server and gdb) part of it for now. We have a board/chip and we are having trouble using it with OpenOCD and PyOCD. Works fine with JLink. If I can create a development/debug flow, then I can also contribute to this nice project and absorb some workload. I have the cycles. Maybe a more exhaustive disassembly debugging :-)

I tried the following

1. The "Launch Extension" config, I can only debug extension.ts
2. The "Debug Server" config, not sure how to use it. It is waiting on a port 4711 not sure what to do after that.
3. The combined of the two above, again I can still only debug etension.ts only.

It appears that the server is running in another process/thread and I seem to be in the wrong one. 

If someone can give me a recipe (or point me in a direction) for debugging this extension, I would be grateful. I will even write up instructions in a markdown document and submit back.

pelrun, 2019-03-28:
Start the combined one. The debug configuration in the child vscode window will also need to be modified, as otherwise it will launch a new debug server rather than use the one connected to the debugger. `"debugServer": 4711` is what you'll need to add.

haneefdm, 2019-03-28:
Many thanks. That was easy!!!

closed 2019-03-28 by haneefdm