Marus/cortex-debug #143: Undefined MI command: undefined

mryndzionek, 2019-03-28:
Here is what happens when I do a restart:
```json
55-interpreter-exec console "monitor reset halt"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"target halted due to debug-request, current mode: Thread \n"}]}
target halted due to debug-request, current mode: Thread 
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"xPSR: 0x01000000 pc: 0x004001ac msp: 0x20458000\n"}]}
xPSR: 0x01000000 pc: 0x004001ac msp: 0x20458000
GDB -> App: {"token":55,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
56-undefined
GDB -> App: {"token":56,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Undefined MI command: undefined"],["code","undefined-command"]]}}
57-thread-list-ids
GDB -> App: {"token":57,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
```

Why is `56-undefined` there? I wasn't able to identify the code that sends this. Could someone point me to the right place? Additionally I've noticed that successive `thread-info` command returns state from before the reset - the pc is not `0x004001ac`.

Another thing I have problems with is debugging `cortex-debug` project itself. Breakpoints for almost all files in the project show up as `Unverified breakpoint`. Has anyone encountered this?

haneefdm, 2019-04-07:
Yes, I saw that 'undefined' command too during a restart. I put it on my list to debug. But have I have a bigger issue with restart. I will try to report back if I find the cause/fix

haneefdm, 2019-04-07:
Perhaps a related issue https://github.com/Marus/cortex-debug/issues/148

haneefdm, 2019-05-02:
@mryndzionek Following code only applies if you are using OpenOCD.

I found a bug in an interaction between OpenOCD and gdb interaction where their caches get out of sync just after reset (especially after a restart). OpenOCD actually has a documented workaround for this problem to make gdb flush its cache. The workaround is to do the following in your startup commands.

```
                /*
                // Following two commands are needed to get gdb and openocd and HW all in sync.
                // Or, execution context (PC, stack, registers, etc.) look like they are from before reset.
                // The stepi, is a pretend instruction that does not actually do a stepi, but MUST be done
                // Its a documented workaround in openocd. Do a 'monitor help' to see more info
                */
                "monitor gdb_sync",
                "stepi"
```

mryndzionek, 2019-05-10:
Thanks for the update. I'm currently developing on a platform where I don't experience the problem, but I'll verify this ASAP.

mryndzionek, 2019-12-17:
I think this issue can be closed. I haven't seen this issue for a while now.

closed 2019-12-17 by mryndzionek