Marus/cortex-debug #145: update pyocd to 0.18.0 cortex-debug can't connect daplink

lgl88911, 2019-03-30:
hi, Marcel：
     I update the pyocd to 0.18.0，and use cortex-debug connect daplink in vscode， It's show "Fail to launch PyOCD GDB Server: Timeout". 
   I use " pyocd-gdbserver --persist -t nrf52 -bh -r" is working well for pyocd 。0.18.0 PYOCD work with my link have a waring "0000201:WARNING:pyusb_v2_backend:Exception getting product string: [Errno 13] Access denied (insufficient permissions)", but 
It doesn't seem to break the CLI work of pyocd.
Could you help check cortex-debug can work well with pyocd 0.18.0, Thanks!

My configure is:
```
        {
            "cwd": "${workspaceRoot}",
            "executable": "/home/frank/work/project/nrf52_moderate/apps/basetest/build/zephyr/zephyr.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "pyocd",
            "targetId":"nrf52",
            "armToolchainPath": "/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/"

        }  
```

haneefdm, 2019-05-10:
Yes, PyOCD changed their output messages which cause us to fail to detect the server starting properly. Fix coming soon.

Marus, 2019-05-15:
This should be fix in V0.2.5. If it is still not working please re-open the issue.

closed 2019-05-15 by Marus