Marus/cortex-debug #147: runToMain does not work most of the time

haneefdm, 2019-04-07:
### Environment:
- OpenOCD with a Cypress PSoC  6 MCU. Kit: CY8CKIT-062-WiFi-BT
- Latest MacOS VSCode and extensions as of today
- Chip is a dual cpu (cm0 + cm4) and what we want to debug is the cm4 (irrelevant I think)
- Bit weird startup that is easily patched via postLaunchCommands in config.

### Problem:
- If 'runToMain' is enabled in the config., it does not work most of the time for this chip. One in 20-30 times it works maybe, if it is not a cloudy day :-)
- A workaround is to disable 'runtoMain' and just burn a manual breakpoint. But we want this feature to work reliably as long as it is there. _**I don't think this is OpenOCD or PSoC 6 specific**_.

It took me almost a week before I understood the problem and implemented a fix.

### Cause: race condition between launch and configuration of breakpoints
- When runToMain is enabled, we do two things during launch
1. Set a temporary breakpoint
2. ask gdb to continue.
- VSCode is still trying to configure existing breakpoints and makes calls even if there are no breakpoints. To handle these requests, since at this point the state looks like !stopped, we **interrupt** exec, reconcile/set those breakpoints and **continue** exec.
- Between the **interrupt** and **continue** execs, you may hit the temp. breakpoint and it will be ignored since **continue** will take precedence and, thus debugger GUI never stops in main()

### My Fix:
- after setting the temp. breakpoint, wait until `configurationDoneRequest()` and then let the target continue.
- You might have a better idea though

### Finally:
I don't know the patch mechanism or if you will at least review an then maybe accept it. New to git. Unfortunately, I did not fork before I made these changes. I can contribute to a few other issues as well. Less than 20 lines of changes perhaps.

Please PM me.

haneefdm, 2019-04-08:
I have my fix on https://github.com/haneefdm/cortex-debug

julian-poidevin, 2019-04-08:
@haneefdm Maybe you should open a pull-request. I'm interested into your fix.

haneefdm, 2019-04-08:
@julian-poidevin I will open a pull request, in a few hours. I want to try it on a Windows box.

julian-poidevin, 2019-04-15:
@Marus : I don't see much activity on the extension repo these last time. Are you the only one who has write rights on master branch ?

haneefdm, 2019-04-15:
I forgot to update this thread. I created a PR a week ago. https://github.com/Marus/cortex-debug/pull/151

haneefdm, 2019-05-07:
@julian-poidevin Could you comment on Issue #162

haneefdm, 2019-05-10:
Closed with PR #151

closed 2019-05-10 by haneefdm