Marus/cortex-debug #153: Cannot start JLink Server debugging

sidprice, 2019-04-11:
HI, I am just trying to get cortex-debug working with JLink Server and cannot get it to start. Here is my launch task:

        {
            "name": "Debug BMP Native ",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "executable": "./blackmagic",
            "showDevDebugOutput": true,
            "interface": "swd",
            "device": "STM32F411RE"
        }

When I try to run my code I see this dialog:

[image]

The path to the JLink Server is:

    "cortex-debug.JLinkGDBServerPath": "C:\\Program Files (x86)\\SEGGER\\JLink_V620i\\JLinkGDBServerCL.exe"

Seems I must be forgetting something, any help much appreciated.

elsonwei, 2019-04-11:
Can you execute arm-none-eabi-gdb directly in the terminal?
If not, you need to set "cortex-debug.armToolchainPath"

sidprice, 2019-04-11:
The path is setup in Windows and this is the output (edited):

PS C:\DataRoot\Projects\blackmagic> arm-none-eabi-gdb --version
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.

elsonwei, 2019-04-12:
Please add `"cwd": "${workspaceRoot}"`

sidprice, 2019-04-12:
Getting closer, now with this launch task:

        {
            "name": "Debug BMP Native ",
            "type": "cortex-debug",
            "request": "launch",
            "cwd": "${workspaceroot}/src",
            "servertype": "jlink",
            "executable": "blackmagic",
            "showDevDebugOutput": true,
            "interface": "swd",
            "device": "STM32F411RE",
            "preLaunchTask": "Make BMP Native"
        }

I get this error from VSCode when I attempt to run the code:

[image]

sidprice, 2019-04-14:
Do you have any suggestions, I really would like to be using your plugin :o(

closed 2019-04-16 by sidprice