Marus/cortex-debug #154: J-Link: Large Peripheral Registers cannot be read

willson556, 2019-04-11:
Using JLink V6.44d with an Infineon XMC4700, examine memory requests > 88 bytes fail:
```
-data-read-memory-bytes 0x40020100 88
{"token":219,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x40020100"],["offset","0x00000000"],["end","0x40020158"],["contents","0700010001000000030000000000000000000000010400000000000000000000000000000000000000000000000000001f1c00001f1c00000000000000000000100e0000100e0000000000003f00000078780000b4b40000"]]]]]}}
-data-read-memory-bytes 0x40020100 92
Unable to read memory. (from data-read-memory-bytes 0x40020100 92)
```
J-Link control panel logs the following:
```
T2180 73088:901 JLINK_ReadMem (0x40020100, 0x0058 Bytes, ...) - Data: 07 00 01 00 01 00 00 00 ...
T2180 73090:394 JLINK_ReadMem (0x40020100, 0x005C Bytes, ...) - Data: AA AA AA AA AA AA AA AA ...
T2180 73090:396 JLINK_ReadMem (0x40020100, 0x0001 Bytes, ...) - Data: AA
T2180 73090:397 JLINK_ReadMem (0x4002015B, 0x0001 Bytes, ...) - Data: AA
```
This results in large peripheral register sets having their values display as all 0. It seems that the read requests need to be broken into smaller chunks.

minkir014, 2019-05-01:
It may be issue with j-link itself.

haneefdm, 2019-05-02:
@willson556 Yes, it could be related to #160. I already fixed the problem I saw and resorted to splitting into 4K chunks. Never imagined something as little as 88 bytes.

I found another issue where if the device is running super slow, then gdb can timeout causing failures.

willson556, 2019-05-03:
Based on the J-Ling log showing garbage data, it seems like it's the thing failing -- not sure why. It seems to work fine in other tools (IAR EW, Ozone) -- not sure what size they're breaking the requests into.

haneefdm, 2019-05-03:
Ummmm that last access to 0x4002015B is unaligned access in the Peripheral area of Cortex and can fail in many devices. From what I know requesting just one byte may also fail (the last two requests) -- needs to be multiple of 4.

We see 4 responses in JLink log right? The first one makes sense. Right address and size. But the next three not so much. Also, we asked for 88 bytes, that somehow turned into 92 bytes by gdb? Since the logs are out of band, could you match the request token with the response token to make sure which response is for which request? Where are the other three requests coming from?

minkir014, 2019-05-10:
@willson556 what's the news?
Does this issue continue?

Marus, 2019-05-15:
@haneefdm has been working hard on code related to this issue, which I've now started looking at with him. Hopefully we'll get something released for this in the near future which should reduce, and hopefully eliminate these issues.

minkir014, 2019-05-15:
As I understood from @haneefdm. The size turned by gdb from 88 to 92. You could search about gdb and why it did that and how other extensions work with that.

Marus, 2019-06-09:
Can you try this on the latest releases which have some changes in how the memory for peripherals are read - may fix the issues you were seeing. Unfortunately I don't have any similar target devices to try to reproduce this on myself.

willson556, 2019-06-09:
@haneefdm: Sorry for the slow response.
> Also, we asked for 88 bytes, that somehow turned into 92 bytes by gdb?

I first asked for 88 bytes to show the request succeeding. The first entry in the J-Link log is for that successful read. The next three entries come from the unsuccessful 92 byte read.

@Marus: Still having the issue with the latest release unfortunately. When inspecting `CCU41_CC40` (at address 0x40010100) on the XMC4700 with the latest release, the following entries are made in the J-Link Log:
```
T3C3C 736:045 JLINK_ReadMem (0x40010100, 0x0040 Bytes, ...) - Data: 00 00 00 00 00 00 00 00 ...
T3C3C 736:046 JLINK_ReadMem (0x40010170, 0x0014 Bytes, ...) - Data: 00 00 00 00 00 00 00 00 ...
T3C3C 736:047 JLINK_ReadMem (0x400101A0, 0x0020 Bytes, ...) - Data: AA AA AA AA AA AA AA AA ...
T3C3C 736:048 JLINK_ReadMem (0x400101A0, 0x0001 Bytes, ...) - Data: AA
T3C3C 736:049 JLINK_ReadMem (0x400101BF, 0x0001 Bytes, ...) - Data: AA
```

In the VSCode Window Log:
```
[2019-06-09 16:13:22.773] [renderer3] [error] Failed  to read peripheral CCU41_CC40:read-failed Error: Unable to read memory: Unable to read memory. (from data-read-memory-bytes 0x400101a0 32)
```

You can find an XMC4700 svd file [here](https://github.com/Infineon/platformio-infineonxmc/blob/master/misc/svd/XMC4700.svd) if that helps debug/understand the behavior. Let me know if there's anything else I can do to help debug this.

haneefdm, 2019-06-10:
I am not using JLink but just parsed the said svd file and here is what I saw (debug output)
```
CCU41_CC40 base=0x40010100, totalLen=256, #ranges=3
**** 0x40010100, len=64, cum-bytes=64
**** 0x40010170, len=20, cum-bytes=84
**** 0x400101a0, len=32, cum-bytes=116
CCU41_CC40 totalLen=256, savings=140 54.69%
```
So, with the new release, there should have been three requests for `CCU41_CC40`. That's it. The request at address `0x400101a0` should have been for 32 bytes and it is. I have no idea where the last two requests are coming from. I doubt they are coming from Cortex-Debug

@willson556 Could you enable debug output for Cortex-Debug in your `launch.json`
```
            "showDevDebugOutput": true,
```
to see what requests are actually being made and if requests are actually failing. You will see a ton out output in the debug console. What is for sure is that the request for 32 bytes at that address is failing with JLink (according to the output you pasted).

I can't reproduce without the hardware.

willson556, 2019-06-10:
Turned on developer debug output. Directly after expanding the `CCU41_CC40` node, the following appears in the debug console:
```
57-data-read-memory-bytes 0x40010100 64
58-data-read-memory-bytes 0x40010170 20
59-data-read-memory-bytes 0x400101a0 32
GDB -> App: {"token":57,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x40010100"],["offset","0x00000000"],["end","0x40010140"],["contents","00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"]]]]]}}
GDB -> App: {"token":58,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x40010170"],["offset","0x00000000"],["end","0x40010184"],["contents","0000000000000000000000000000000000000000"]]]]]}}
GDB -> App: {"token":59,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Unable to read memory."]]}}
```

J-Link log shows:
```
T2A54 248:968 JLINK_ReadMem (0x40010100, 0x0040 Bytes, ...) - Data: 00 00 00 00 00 00 00 00 ...
T2A54 248:970 JLINK_ReadMem (0x40010170, 0x0014 Bytes, ...) - Data: 00 00 00 00 00 00 00 00 ...
T2A54 248:971 JLINK_ReadMem (0x400101A0, 0x0020 Bytes, ...) - Data: AA AA AA AA AA AA AA AA ...
T2A54 248:972 JLINK_ReadMem (0x400101A0, 0x0001 Bytes, ...) - Data: AA
T2A54 248:973 JLINK_ReadMem (0x400101BF, 0x0001 Bytes, ...) - Data: AA
```

willson556, 2019-06-10:
Just tried using Segger Ozone with the same SVD file and it did an individual request for each register:
```
T3EC4 252:235 JLINK_ReadMemU32(0x40010100, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:237 JLINK_ReadMemU32(0x40010104, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:238 JLINK_ReadMemU32(0x40010108, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:239 JLINK_ReadMemU32(0x4001010C, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:240 JLINK_ReadMemU32(0x40010110, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:241 JLINK_ReadMemU32(0x40010114, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:243 JLINK_ReadMemU32(0x40010118, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:244 JLINK_ReadMemU32(0x4001011C, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:245 JLINK_ReadMemU32(0x40010120, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:247 JLINK_ReadMemU32(0x40010124, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:248 JLINK_ReadMemU32(0x40010128, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:249 JLINK_ReadMemU32(0x4001012C, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:250 JLINK_ReadMemU32(0x40010130, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:250 JLINK_ReadMemU32(0x40010134, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:251 JLINK_ReadMemU32(0x40010138, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:252 JLINK_ReadMemU32(0x4001013C, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:253 JLINK_ReadMemU32(0x40010170, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:255 JLINK_ReadMemU32(0x400101A0, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:256 JLINK_ReadMemU32(0x400101A4, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:258 JLINK_ReadMemU32(0x400101A8, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:259 JLINK_ReadMemU32(0x400101AC, 0x0001 Items, ...) - Data: 00 00 00 00
T3EC4 252:260 JLINK_ReadMemU32(0x400101B0, 0x0001 Items, ...) - Data: 00 00 00 00
```

haneefdm, 2019-06-10:
@willson556 @Marus Ok, I may have found the issue. **Maybe**

In that set of registers, registers `SWS/SWR`, all fields are marked as `write-only` while the registers themselves are not marked as such. Which is why a read is probably failing. I can make a change so that we skip such registers (`iff` all fields are write-only). But, this does not sound normal for ARM implementations that I know of.

@willson556 Would you be able to get a change from me to test? I want @Marus to chime in as well.

willson556, 2019-06-10:
Good spot! Yeah, I can checkout and build/run a version of the extension from your GitHub to test with.

Thanks for your help!

haneefdm, 2019-06-10:
@willson556 Hmmm, in the Ozone, it did read `SRS (0xa8)` and `SWS (0xac)` and those reads did succeed?!?! So, my theory might not be right. Tomorrow, I will modify code regardless and see if we can still do bulk reads. Even Eclipse does bulk reads. Not being able to do bulk reads will be bad.

haneefdm, 2019-06-10:
@willson556 Could we try something before we make any changes to the code. Lets prototype by modifying the svd file. Make a copy and remove SWS/SWR registers from the copy, point your launch.json to the modified file and see if those particular read errors go away.

Delete lines 71569-71700.

Note that there many other registers like these in other peripherals, but we could just focus on these two.

haneefdm, 2020-03-30:
I believe this issue is resolved. Please re-open if needed

closed 2020-03-30 by haneefdm