Marus/cortex-debug #155: Feature Request: Formalize toolchain and server paths customizable by platform

haneefdm, 2019-04-12:
I believe the code is already there to do everything needed. But it is not declared in package.json schema and I believe there is also an error for `armToolchainPath` in there as well.
```
            "osx": {
                "toolchainPath": "/Applications/ModusToolbox_2.0/tools/gcc-7.2.1-1.0/bin",
                "serverpath": "/Applications/ModusToolbox_2.0/tools/openocd-2.2/bin/openocd",
                "armToolchainPath": "xxxyaaz"  // Has no effect
            },
```
The `toolchainPath` and `serverpath` actually do work but not specified in package.json and Intellisense. **Could we formalize these features please**?

We have team members working on multiple OSes and switching between them, being able to switch easily can be really helpful. settings.json (local or users home dir) does not have OS specific overrides (I think).

It is unfortunate that `serverpath` is not camelCase and neither is `toolchainPath` strictly speaking, so we have to stick with them for compatibility.

The `armToolchainPath` property shows up in IntelliSense but has no effect in launch.json; it is really for settings.json? I see in changelog v0.1.20 that a change was made to support per-launch basis. Not seeing evidence of that.

# Fixes?
I can fix this by editing the package.json. Don't want to rename config. names though unless there is a consensus. I volunteer to help with the approval of the author/team. Please comment.

minkir014, 2019-05-01:
You can use Microsoft cpp extension for intellisense. I can help you with this. And use this extension for debugging.

haneefdm, 2019-05-01:
@minkir014, I already use the Microsoft cpp extension and also providing bug fixes to those repositories (MIEngine and vscode-cpptools) as well. This is not about editing C/C++ code. I am talking about when editing launch.json for my embedded c++ project for the `cortex-debug` debugger configuration.

I debug using both extensions (vscode-cpptools and cortex-debug)

minkir014, 2019-05-01:
If you have the solution to fix that. Start a PR.

Marus, 2019-05-16:
@haneefdm - any option in launch.json can be put in an OS specific block already - and you should be getting auto-completion. The options that are in your user/workspace settings don't have that - as those types of settings are generally specific to the machine and not the project. 

I wasn't originally intending for the the various gdb server paths and stuff to be configured in the `launch.json` file, they were intended to be set through user/workspace settings as I considered those to be more machine configuration specific than project specific (armToolchainPath was exposed intentionally for one particular use case - although I'm still not actually convinced it was a good - I still think it would have been better to be using workspace settings for that). 

The reason why the `serverpath` and `toolchainPath` work is more accidental than intentional - while I didn't expose them in the package.json - they happen to match up with the internal variable names that the user/workspace settings get mapped into - and I didn't explicitly strip them from the config values that we get from the launch.json file.

Also - if `toolchainPath` is not set in the launch.json then the exposed `armToolchainPath` value should be working.

And while this isn't significant - I would argue that the camel casing on armToolchainPath is correct - toolchain is generally considered to be one word.

haneefdm, 2019-05-16:
@Marus the bigger thing is being able to set them on OS basis. Multiple users on a given project can make them relative to some environment variable we can still use `${env:HOME}/...`

Workspace settings get checked into version control and users have to constantly mess with it. **Looks like user setting is the only safe place then?**

You are right `armToolchainPath` and `toolchainPath` are already camel case.

> Also - if toolchainPath is not set in the launch.json then the exposed armToolchainPath value should be working.

Ahhh okay.

Just wanting to see what should be formalized and work with IntelliSense. When starting out, IntelliSense is your main guide to figure out what options are supported and what they mean.

I didn't see it as a bug/issue per se. but got confused with what works where and why. Hence marked it as 'Feature Request'.

closed 2019-05-16 by haneefdm