Marus/cortex-debug #156: Random crashes in the debug adapter

haneefdm, 2019-04-13:
It happens when I stop a debug session once in a blue moon (launch or attach does not matter). Very difficult to reproduce, but it kills the debug adapter. I am using the adapter in server mode (debugging stuff). This is the stack trace you see...

```
Error: kill ESRCH
    at process.kill (internal/process/per_thread.js:198:13)
    at Timeout.stop.setTimeout [as _onTimeout] (/Users/hdm/cypress/vscode/cortex-debug/src/backend/mi2/mi2.ts:224:47)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```
I have not investigated it much but it may be a condition where the server/gdb already quit and we are trying to kill it. Just wanted to capture it as an issue.

How can I help debug it? Or see how this can even happen. If this is a known issue please let me know. I can help in whatever way you want.

minkir014, 2019-05-01:
This happens when you launch or when you stop the debugger.

haneefdm, 2019-05-01:
When I hit the stop debugging button (ending a debug session). The button next to restart. From what In the source code, we ask gdb to exit and then for good measure, a second later also try to send a kill signal if not already quit. There may be a race condition between the three lines of code at the end of stop/detach. In src/backend/mi2.ts. A try/catch around process.kill() may be necessary
```
    public stop() {
        const proc = this.process;
        const to = setTimeout(() => { process.kill(-proc.pid); }, 1000);
        this.process.on('exit', (code) => { clearTimeout(to); });
        this.sendRaw('-gdb-exit');
    }

    public detach() {
        const proc = this.process;
        const to = setTimeout(() => { process.kill(-proc.pid); }, 1000);
        this.process.on('exit', (code) => { clearTimeout(to); });
        this.sendRaw('-target-detach');
    }
```
When the gdb adapter dies, future debugging sessions will stop working.

minkir014, 2019-05-01:
You can start a pull request as you started before that and I would support you. Make try/catch expressions as you said and try it and tell me the result.

haneefdm, 2019-05-01:
@minkir014, I already have a PR #151  in the queue from more than 3 weeks ago. I am trying not to do additional PRs until that one is accepted. I am trying to document issues I find to go back and fix one at a time. I know the fixes for almost everything I find but also helps to get feedback.

My next PR will be fixes for SVD handling (Issue #160) in certain cases to avoid a bug I found in OpenOCD. Also submitted a PR in the OpenOCD repo.

minkir014, 2019-05-01:
Have you tried your solution?
Why don't you make a fork of this project and be the responsible for it.

haneefdm, 2019-05-01:
@minkir014 I don't want to just coopt @Marus work. Don't feel right about it. Give him a bit more time.

I think I also have to create a new name for the extension and the debugger to get approved by MS/Marketplace

minkir014, 2019-05-01:
These things are very easy. Choosing a new name and publishing it to the marketplace won't be hard.

minkir014, 2019-05-01:
Also have a look at this issue. https://github.com/Marus/cortex-debug/issues/158

haneefdm, 2019-05-01:
@minkir014, I am not talking about hard. I am talking about giving due respect to @Marus. Maybe he is going through a personal/health/work issue. I have been there.

@Marus could give me write permission for example and withdraw at a later date. Two almost identical extensions will also confuse users. Mine will have very few downloads ... which one will new users go for and how will existing users know about the new extension. Creates confusion, I think. But, in a week, I may change my mind.

minkir014, 2019-05-10:
@haneefdm look with me in these issues #117 and #120

haneefdm, 2019-05-10:
I think they are unrelated

minkir014, 2019-05-10:
I want you to help me make this extension better. So, Can you look at those issues?

haneefdm, 2019-05-10:
@minkir014, I did look at the issues you mentioned and responded

Marus is back and he added me as a collaborator, which is super nice. Slow down though, we have to be very careful to figure out what is actually wrong and find an appropriate fix. Or we will break everything and the extension will become useless. You can be more helpful if you analyze issues carefully or even learning the code for this extension.

haneefdm, 2019-06-23:
I haven't seen a crash in this area in a long long time since I added the try/catch

closed 2019-06-23 by haneefdm