Marus/cortex-debug #158: Can't launch my program on external DDR

teramil, 2019-04-18:
It's my launch.json:
```json
 {
           "name": "EA-16V Download Debug",
            "type": "cortex-debug",
            "request": "launch",
            "executable": "${workspaceFolder}/build/program.elf",
            "servertype": "jlink",
            "interface": "jtag",
            "armToolchainPath": "C:/arm-none-eabi-gcc-4_7_3/bin",
            "device": "MK61FX512xxx12 (allow security)",
            "cwd": "${workspaceFolder}",
            "runToMain": true,
}
```

When I start debugging I see this:
[image]
If I make a step by debug I see this:
[image]

In internal flash I have bootloader. Bootloader copy my program from flash mamory to DDR. I need to start debug session on DDR in launch mode.

Attach mode works fine.

minkir014, 2019-05-01:
I didn't understand your problem correctly. When you launch the debugger it works but when you make a step it freezes. Am I right? Or no.

haneefdm, 2019-05-01:
I think I can understand what is happening. Post bootloading, the program is relocated and if someone (gdbserver, gdb, cortex-debug) is caching, symbol/address info. Post relocation, source code mapping is lost. The program has to be compiled and linked (linker script) properly as well and I assume you did everything right, because you are able to attach and run and source code mappings work.

This is a very common usage scenario. I will put it in my queue to look at what could be happening.

Here is an app-note http://www.keil.com/appnotes/files/apnt_138.pdf in case your linker script is not fully specified. Also see. http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka9466.html

If you are already doing everything right, then please and still having issues, please post back.

minkir014, 2019-05-01:
Do you have any reasons that attaching work but launching don't?

minkir014, 2019-05-01:
https://github.com/teramil please answer if you want us to solve this problem.

haneefdm, 2019-05-01:
@minkir014  Like I said if some SW may be caching the original addresses before the bootloader ran then results after could look wrong. Just an informed guess at this point ... not knowing more details.

**Rambling below:**

I found one OpenOCD case where gdb and openocd were caching some info and after a restart, stack trace/pc was wrong, thread info was wrong, etc. I don't blame gdb because it was never notified of that change. I had to flush that cache by issuing a fake 'stepi' command. Supported by OpenOCD as a workaround and takes two gdb commands.
```
    "monitor gdb_sync",   // Instructs OpenOCD to ignore the next stepi
    "stepi"               // This gives gdb a new context and flushes all it's caches.
```
This issue can be super important because it is a very common use case. Sometimes we just move interrupt tables and routines themselves into RAM to get better ISR response times.

Btw, I don't claim to be a super-duper expert at all of this. Too many moving parts. Just based on my experiences and researching what is there. I welcome your feedback though @minkir014

minkir014, 2019-05-01:
So you think that the problem is in gdb.

haneefdm, 2019-05-01:
I don't know @minkir014 Not enough info to diagnose. I want to make sure the program was compiled and linked properly. What do you think?

minkir014, 2019-05-01:
You may start with starting an issue on gdb site and then decide where is the error

minkir014, 2019-05-02:
Any news about this issue @haneefdm ?

haneefdm, 2019-05-02:
@minkir014 I can't look at at this until @teramil responds to make sure he is compiling/linking properly according to how ARM tells you to do. I don't know what to look at? Is there a sample project @teramil can share?

haneefdm, 2019-05-02:
@teramil, when you single step, I see that your PC has moved somewhere else but it still looks like it is in FLASH/code area (0x00000000-0x20000000). If you are running from eternal RAM, shouldn't it be in the range of 0x60000000-0xA0000000

What happens when you do a continue and then break instead of step?

teramil, 2019-05-06:
> I didn't understand your problem correctly. When you launch the debugger it works but when you make a step it freezes. Am I right? Or no.

No, when I make a step debugger was switched from DDR to internal RAM bootloader's area. It must step over on DDR and stop in main function, but I see other things.

teramil, 2019-05-06:
@haneefdm yeah debugger must write actual version of my program to external DDR area and run it from there without bootloadiing. In my first picture you see PC point to needded area but when I make a step PC point to bootloader's area I have no idea why.

minkir014, 2019-05-06:
Was your compiled and linked (linker script) properly before you made the step debugger?

haneefdm, 2019-05-06:
@teramil, I could be wrong but I don't think gdb will move your program. If it did, you will see that in the JLink output or [log file](https://wiki.segger.com/Enable_J-Link_log_file). The linker can generate code to do that as part of the startup code, which will run out of FLASH

```
.text: { ... } >ram AT>rom
```
Now, gdb will know where your code is stored vs where it is running. Do you mind posting your linker script? And, the compiler options you are using? It may also be important to compile with relative addresses so that func. calls do not go back to flash. I am sure you are doing all of that properly, but I can't duplicate the problem or help if don't see the compile and link settings.

[Here is an article
](https://stackoverflow.com/questions/15137214/how-to-run-code-from-ram-on-arm-architecture) that may be helpful. Also, generate and study the map file to make sure the linker is doing everything right.

teramil, 2019-05-07:
Here's the part of my linker script file: 

```
    m_boot_info       (RW) : ORIGIN = 0x0000a000, LENGTH = 0x00001000        /* internal ROM: boot information */
    m_kernel_data     (RW) : ORIGIN = 0x1fff0000, LENGTH = 0x0001f000        /* internal RAM: kernel space (Location of MQX Kernel data + MQX heap) */
    m_bstack          (RW) : ORIGIN = 0x2000fa00, LENGTH = 0x00000500        /* internal RAM: Boot stack reused by MQX Kernel data */
    m_start_info      (RW) : ORIGIN = 0x2000ff00, LENGTH = 0x00000100        /* internal RAM: start info section */
    m_firmwareinfo    (RW) : ORIGIN = 0x08000000, LENGTH = 0x00001000        /* external DDR: firmware information */
    m_vector_table    (RX) : ORIGIN = 0x08001000, LENGTH = 0x00000400        /* external DDR: interrupt table */
    m_text            (RW) : ORIGIN = 0x08001400, LENGTH = 0x001fec00        /* external DDR: text */
    m_data            (RW) : ORIGIN = 0x08200000, LENGTH = 0x00100000        /* external DDR: data */
    m_journal_gs      (RW) : ORIGIN = 0x80300000, LENGTH = 0x00010000        /* external DDR: journal gs */
    m_snapshot        (RW) : ORIGIN = 0x804ff000, LENGTH = 0x00001000        /* external DDR: system snapshot */
    m_app_heap        (RW) : ORIGIN = 0x80500000, LENGTH = 0x01b00000        /* external DDR: data */
```

I think linker does everithing right because if I run my program without debugging I see following:
1. Bootloader's starting and load my program from external flash to DDR
2. Bootloader starts my program from DDR
2. My program executes correctly 

The problem described above is in only debug mode.

minkir014, 2019-05-07:
Have you verified that your program has been successfully compiled and linked?

teramil, 2019-05-07:
@minkir014  sorry I don't undestand how I can verify it. What do you mean?

haneefdm, 2019-05-07:
@teramil, in your case, I don't think you can skip steps 1 and 2. Gdb may issue commands to jlink to copy the program/data to the said addresses, in the program load phase right after a reset. But how will jlink do that? Your external memory is still not initialized/setup by the processor. Don't you have to initialize some registers, and setup the DDR before anyone can use it?

It is interesting that your program is in 0x0800XXXX address which I thought was still in flash. [See cortex memory map](https://developer.arm.com/docs/ddi0337/e/memory-map/about-the-memory-map) Your device may do something different.

I would run everything from the command-line because I don't think this has anything to do with this extension. All this extension does is to setup the device, set a break point at main and just wait for that (or any) break-point to be hit. You can enable `"showDevDebugOutput": true` in your launch.json,  and look in the debug console of VSCode. You will see every step of what is going on. Takes some patience.

You can also switch to the OUTPUT tab and see the "Adapter Output" only available in the dropdown.

[image]

minkir014, 2019-05-07:
But why it runs from command line and not working in this extension?

haneefdm, 2019-05-07:
He never said it runs fine from the command line. He said it runs fine when
he attaches (instead of launch) to a running program which means the
bootloader did its job of moving the program. When you attach, there is no
program load, gdb just uses whatever is already running.

teramil, 2019-05-07:
@haneefdm when I power on my device bootloader is started and it intializes my ddr and clock registers. Can debugger reset my ddr and clock settings? I don't know...

haneefdm, 2019-05-07:
@teramil Nope, gdb cannot do that and neither can JLink or any other gdb-server.

I think we found your problem. You cannot skip your startup code that does your magic. It will eventually get to main(). The way you setup your linker script is not how I would have done it.

haneefdm, 2019-05-07:
@teramil you might want to close this issue as it has nothing to do with this extension

teramil, 2019-05-08:
@haneefdm Why should I skip startup code? And what's happend if I don't skip startup code? Explain please or give me links to articles. Thank you very much for your help!

closed 2019-05-08 by teramil

haneefdm, 2019-05-08:
During a launch, a hard reset is done and I don't know about your board/processor but your RAM contents are probably gone and all peripherals (ddr clock/registers) will probably be gone. One way or another, you have to get your RAM working after a reset. You should have the right jump to entry point in RAM through the reset vector which is not in RAM initially which probably starts your boot loader.

teramil, 2019-05-13:
@haneefdm Can I use tcl init files for kinetis mcu with the extention?

haneefdm, 2019-05-13:
@teramil What are you trying to do?

Tcl is really a feature of the gdb-server like OpenOCD and not this extension but this extension allows you to pass tcl files to OpenOCD.

minkir014, 2019-05-13:
Until this extension support that platformio extension support TCL.