Marus/cortex-debug #159: Add to watch list fills in wrong variable name (in a different way)

adammunich, 2019-04-18:
In the watch list, when inspecting a variable, "public" is always a member of the tree. 

So you would see... 

**Grandfather**.public.**Parent**.public.**Child**

Adding the variable to the watch list would cause an invalid expression error unless the .public's were removed manually. Changing one line in my gdb.js fixes this, from:

const varObjName = `watch_${exp.replace(/\./g, '__').replace(/\[/g, '_').replace(/\]/g, '_')}`;

to:

const varObjName = `watch_${exp.replace(/\./g, '__').replace(/\[/g, '_').replace(/\]/g, '_').replace(/\.(public)/g, '')}`

Marus, 2019-05-15:
I think this should be fixed by release v0.2.6 - which should be available in the marketplace shortly. Please confirm this fixes your issue.

Marus, 2019-05-16:
I'm going to mark this as closed - if you still have an issue please reopen the ticket.

closed 2019-05-16 by Marus