Marus/cortex-debug #160: Peripheral updates have silent failures with OpenOCD

haneefdm, 2019-04-21:
Almost all peripherals show all zeros for this device svd. I don't think it is actually a defect in Cortex-Debug but a combination of things give false results to end user and Cortex-Debug could be more efficient.

**Problem. Bit difficult to explain.**
* The Cypress PSoC6 device svd file specifies that the size of each peripheral is 64K or 128K when in fact very little of that space is used.
* Cortex-Debug issues a memory-read command to gdb for the full size for a given peripheral in an expanded state
* GDB is instructed by OpenOCD that it can only handle (8K-1) as the largest request. **This is a bug in OpenOCD** [Patch has been submitted to OpenOCD](http://openocd.zylin.com/#/c/5109/) by my colleague. Not sure when it would be integrated
* So, GDB splits the request into multiple (8K-1) reads which cause unaligned reads and reads fail.
* On failure, OpenOCD returns all zeros and causes a silent failure downstream. Neither GDB nor Cortex-Debug knows that read has failed. You can see the reads fail if OpenOCD is invoked with -d3 option in the adapter output window.

**There are two issues here**
* The obvious one is the OpenOCD bug which I already tried to address
* Cortex-Debug should not be issuing such large memory read requests regardless of which gdb-server we are using. Eclipse SVD plugin does much better. What we do is very inefficient; takes a while for the data transfer to occur. I believe a peripheral can also have buffers that are included in the size. We should be selective on what we want to read.

**Fixes**
* I fixed it by doing a mark and scan of the actual addresses in use and do a gather and merge address ranges that are close enough. Merge is optional.

The merge produces half as many gdb requests as Eclipse for about the same number of bytes. Data transfer goes from 64K down to about 1K. But, we now have 17 read requests instead of 1. Eclipse does twice as many reads compared to mine.

**My reservations; HELP needed, Please PM me**
While my fix works, **I don't like it**. The mark/scan/gather/merge is all fine (in isolated files) and I have unit tests for them. I just don't know how to issue a set of async memory reads and wait on all of them to finish before updating the registers. My workaround was to `await` on all the read requests. That's the part I hate. We are supposed to play nice

Tried many things like `Promise.all()` and it was not working.

I don't want to do a pull request just yet until I get some advice. @Marus will probably kill me if I did a PR right now :-)

haneefdm, 2019-04-21:
@julian-poidevin Please comment if you have time

haneefdm, 2019-04-22:
I now have a version which does not do any reduction of address ranges, but it will still split up anything larger than 4KB into multiple requests to work around the OpenOCD bug.

The difference in windows refreshing is HUGE. 2+ seconds for reading a whole 64K vs, reading smaller optimized chunks of actually used addresses is barely noticeable. Just for a single step. Less than 250ms, no one will notice. But 2-3 seconds for every action, not nice.

Note, I am looking at what the user is going to see, vs how many CPU cycles on my laptop

haneefdm, 2019-05-02:
I could not duplicate it but gdb can time out on a read. Reading large memory spaces on a slow running device can cause failures. I think you can set the timeout limit in gdb but most people will not know how to do that. Splitting into reasonably smaller chunks still makes sense.

willson556, 2019-05-02:
This seems similar to #154 (which is with Jlink + GDB).

haneefdm, 2019-05-28:
Should be addressed in the next release.

Marus, 2019-06-14:
@haneefdm - have you tried this out since the latest release to see if that has fixed the issues? Just wondering if we can close this ticket or if we still need more work here?

haneefdm, 2019-06-14:
Yes, this is fixed and I verified it in 0.2.7

closed 2019-06-14 by haneefdm