Marus/cortex-debug #161: Breakpoints working but cannot step

BEmbedded, 2019-04-23:
Hi Marus,

I'm using a J-Link to program an STM32 and this plugin for VSCode has normally worked perfectly. However, I've recently noticed an issue where breakpoints work perfectly but I can no longer step in or over. 

I can set a breakpoints on the next line and resume to it fine but pressing the step in or step over buttons have no effect and don't show anything in the debug console.

Any idea what might be causing this?

noppej, 2019-04-24:
@BEmbedded  ... I experience the same thing, but it is intermittent. Once it starts, the only way to clear it, is to Debug->Remove All Breakpoints, and then set just this one breakpoint.  Is the behaviour similar for you?

BEmbedded, 2019-05-01:
@noppej Sorry for the late reply. Mine is also intermittent (nearly always not working) but I am unable to fix it like you described. I had it working perfectly once yesterday and then I reset the program and its been broken again ever since.

If it makes any difference I'm running on an up to date Arch Linux system.

BEmbedded, 2019-05-02:
I have investigated this problem some more and made the following discoveries:
* Debug builds were debugging fine but not Release builds.
* Adding ``` "runToMain": true, ``` to my launch configuration appears to fix debugging for Release builds.

I'm unsure why runToMain would help my debugging issue, but hopefully this helps others with the same issue.

noppej, 2019-05-02:
I'm running on MacOS so that might explain the difference. Do you use an RTOS? I know that when openOCD launch is configured to recognize an RTOS, then there are some known problems with breakpoints that occur before the RTOS kernel is initialized (this would include a breakpoint caused by runToMain in debug mode).

haneefdm, 2019-05-07:
@BEmbedded I think one difference, normally, the program stops at the first instruction (reset vector). When 'runToMain' is enabled, Cortex-Debug sets a temporary break-point in main and issues a continue.

Can't explain why Debug builds run fine and not Release builds ... unless you are using an RTOS like @noppej said. Also, why intermittent?

Marus, 2019-05-20:
In general release builds tend to have a lot more optimizations enabled, and quite often, have debug symbols stripped or not generated. This will often cause problems with debugging the build - not just with cortex-debug but with any debugger. 

Can you post the flags you are using for GCC when compiling - may be able to tell a bit more about what is going to be possible with that.

mcrajah, 2019-07-12:
I face a similar issue where the debug floating bar step in, step over etc dont work, however setting a break point and clicking the green run button works. In addition to this, in the Debug Console window, if I use GDB commands like s, n, si, ni, they work fine, issue is limited to the floating debug bar buttons only. 
Others facing the same issue can use the GDB command work around mentioned above till a root cause is identified.

haneefdm, 2019-07-12:
Okay, **maaaybe** I see what is going on. I have the following situation (and maybe you do too). I am stopped at a point where I have the following situation:
- I do **not** have `Forced Disassembly` mode turned on (it is in Auto mode)
- my top of the stack is in a function with no source code and maybe more such frames
- the debugger normally displays the first frame with source code

You now do a `step`/`next` using the debug toolbar. What the debugger is actually doing is a `stepi`/`nexti` for the top of the frame which is machine instructions (hidden) -- so it is stepping through the instructions at the top of the frame.

To confirm this, look at the address in the call-stack window for the top of the frame; Better yet, add `$pc` in the Watch window. You will notice that the PC is actually moving but the cursor in the C editor window seems stuck.

Could anyone confirm that this is what you are seeing?

As for my situation, this should be fixed and I will try to do that after consulting with @Marus

haneefdm, 2019-07-12:
Oh, there is one more caveat in the situation I described above. If the top of the frame source code is not available but if that source was compiled with debug symbols/lines then even doing a 'next'/'step' from debug console will still stop at the next source code line in some unknown source. No good way to fix that situation. Only step-out will work.

haneefdm, 2020-03-30:
Hi @BEmbedded, @noppej Could you please try the latest release and see if there is any relief? Please re-open if it is still a problem

closed 2020-03-30 by haneefdm