Marus/cortex-debug #162: Requesting write access to this repo. Letter to @Marus or Marcel

haneefdm, 2019-05-02:
Hi @Marus, I don't know how to contact you other than through this forum. You can send me a private email @haneefdm and we can talk offline.

I completely understand that you are unable to attend to this repo in a while (maybe 2-3 months). I have been there before due to family and/or health issues. God forbid nothing bad happened to you or your loved ones. _Hope you are just busy, taking a long vacation, doing something fun, something I wish I should be doing, etc._

I will promise to be super careful in merging changes, etc. I can watch over this repo until you get back. I may not be as good as you but competent enough. I think longer term, worry about impact, design, changes & features that stick. Of course, I can't publish new releases via the marketplace but I can provide vsix files that are easy enough to install. Pull by users rather than Push.

When you get back, you can rescind my write access or think of me as a valuable contributor. I suggest you would want collaborators on a project like this -- me or others.

If I don't hear from you **(gnashing my teeth typing this)**, do you mind that I fork this project and create a new extension? I HATE to do this because it causes confusion in user-land when you have two extensions and the latest extension would have near zero downloads or reviews. I think, creating a new extension is the worst outcome for **everyone**.

To the community, is there a better way? What do you want me to do? A few people have already said I should fork but this is emotional to me. I feel like I am stealing something.

noppej, 2019-05-02:
@haneefdm I sincerely hope that Marcel will see how sincere you are in your intentions, and will support your efforts. He has created a very valuable extension, but as we all know, projects loose value over time, unless tended by creators or other interested surrogates. I am grateful for your efforts, and like you, I hope that Marcel has been distracted for positive reasons. You have my full support.

diegoherranz, 2019-05-02:
I agree on the confusion created if this is forked, but on the other hand that is the good thing about open source.

The key for projects to succeed and survive is having more than one person behind it, so I think it would be very beneficial even if @Marus is back.

Thank you all, especially @Marus, for this nice project!

haneefdm, 2019-05-03:
To add one more thing. Before posting this letter, I went through 50+ issues submitted here to understand the thinking behind what @Marus fixes, what he defers or rejects and why. I am in agreement with 90+% of his decisions. I understand his rationale and philosophy, I think. I would have done the same. How one fixes issues may be different but that is nit-picking. I've been lurking since last Nov. and studying the source code for 2-3  months.

I know when to ask for technical help and hopefully, I will have volunteers. Since we are working with HW, logistical help is also needed.

danielRep, 2019-05-06:
After the readiness and kind help of @haneefdm to solve my issue (#163) , and after reading this letter, I totally support him.

nergnezor, 2019-05-06:
Do it, @haneefdm !

haneefdm, 2019-05-06:
Procedurally, I am thinking, the following may be workable
- I provide my repo as a fork of this extension but do not change name, owner and such. Will not create a new extension. When @Marus gets back, he can choose what he wants to do
- Provide updates in the form of vsix files -- easy to download and to install run a command `code --install-extension cortex-debug-<version>.vsix`. You can do it from the VSCode GUI as well. The downside, **you will NOT get notified** unless you are watching my repo for all changes. Maybe there is a better way. Maybe another repo that just has vsix downloads? GitHub has a feature to only watch releases, not sure how it works.
- Each vsix release will have a version number and documents what issues were addressed
- Users have to file issues on **my** repo rather than this one and I hope I will have some volunteers to triage the issues. I don't like this method because how would users know to do that? Or I have to monitor this repo constantly. **Ideas, people?**
- About pull requests. If you fix something, you **have** to do a pull request against my repo. I will figure out what to do with existing pull requests.

Nothing about this ideal. I am trying to get a temporary solution in place.

danielRep, 2019-05-07:
In my opinion you can have the fork and another repository with only the .vsix downloads. Anyone that wants to follow in more detail the changes can follow the fork, and someone that wants directly to download an update version of the plugin goes to the other repository.

julian-poidevin, 2019-05-07:
I support @haneefdm at 100%. I think he can totally either take the hand on this repo or create a fork of it. I think we should all find a way to contact @Marus. Maybe using social network? Would be really great to have his opinion on this.

I'm really looking forward improving and contributing to this repo since I use it regularly (at least once per week) at my workplace and it helped me A LOT.

haneefdm, 2019-05-09:
Update: My boss just made contact with Marcel. He is doing well, just busy. I am glad we did not do anything yet and perhaps I won't have to. Please be patient until he chimes back into this repo.

haneefdm, 2019-05-10:
He is back!!!

closed 2019-05-10 by haneefdm