Marus/cortex-debug #163: Debugging Arm Musca-A platform with PyOCD timeout

danielRep, 2019-05-06:
Hi,

Recently I started setting up an environment do develop/debug Arm Musca-A board on a Ubuntu 18.04 machine. 

After installing pyOCD, I can run directly on the command line and open a gdb session with a stable connection (i can issue load, breakpoint, continue commands without running to any issues). This seems stable and working fine.

When I try to use pyOCD with Cortex-Debug the vscode keeps showing an error: "Failed to launch PyOCD GDB Server: Timeout".
My launch.json configuration file looks like this: 
```
{
            "cwd": "${workspaceRoot}",
            "executable": "./leds.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "pyocd"
 }  
```
Simple but enough following the Launch Configuration guide, since I don't have any other board attached so boardId is omitted. I also tried to pass a svd file (svdFile) which I got from the Keil packages w/o any effect. The Adapter Output console shows that gdbserver server was been started. 

> 0000429:WARNING:gdb_server:pyocd-gdbserver is deprecated; please use the new combined pyocd tool.
0000526:INFO:board:Target type is musca_a1
0000638:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0000700:INFO:ap:AP#0 IDR = 0x54770002 (APB-AP var0 rev5)
0000730:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0000760:INFO:ap:AP#2 IDR = 0x84770001 (AHB-AP var0 rev8)
0000796:INFO:rom_table:AP#0 ROM table #0 @ 0xf0000000 (designer=43b part=797)
0000814:INFO:rom_table:[0]<f0001000:CSTF class=9 designer=43b part=908 devtype=12 archid=0000 devid=0:0:34>
0000829:INFO:rom_table:[1]<f0002000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0000842:INFO:rom_table:[2]<f0080000:ROM class=1 designer=43b part=798>
0000843:INFO:rom_table:  AP#0 ROM table #1 @ 0xf0080000 (designer=43b part=798)
0000860:INFO:rom_table:  [0]<f0081000:TPIU class=9 designer=43b part=912 devtype=11 archid=0000 devid=0:0:a0>
0000874:INFO:rom_table:  [1]<f0082000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0000886:INFO:rom_table:  [2]<f0083000:TSGEN class=15 designer=43b part=101>
0000906:INFO:rom_table:AP#1 ROM table #0 @ 0xf0008000 (designer=43b part=744)
0000920:INFO:rom_table:[0]<f0009000:GPR class=9 designer=43b part=9a4 devtype=34 archid=0a34 devid=0:0:1>
0000924:INFO:rom_table:Enabled power to power domain #0
0000933:INFO:rom_table:[1]<e00ff000:ROM class=1 designer=43b part=4c9 pwrid=0>
0000934:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0000948:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0000958:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0000967:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0000978:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0000986:INFO:rom_table:  [5]<e0041000:ETM class=9 designer=43b part=d21 devtype=13 archid=4a13 devid=0:0:0>
0000994:INFO:rom_table:  [6]<e0042000:CTI class=9 designer=43b part=d21 devtype=14 archid=1a14 devid=0:0:40800>
0001012:INFO:rom_table:AP#2 ROM table #0 @ 0xf0008000 (designer=43b part=744)
0001028:INFO:rom_table:[0]<f0009000:GPR class=9 designer=43b part=9a4 devtype=34 archid=0a34 devid=0:0:1>
0001032:INFO:rom_table:Enabled power to power domain #0
0001041:INFO:rom_table:[1]<e00ff000:ROM class=1 designer=43b part=4c9 pwrid=0>
0001041:INFO:rom_table:  AP#2 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0001056:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0001064:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0001074:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0001084:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0001094:INFO:rom_table:  [5]<e0041000:ETM class=9 designer=43b part=d21 devtype=13 archid=4a13 devid=0:0:0>
0001103:INFO:rom_table:  [6]<e0042000:CTI class=9 designer=43b part=d21 devtype=14 archid=1a14 devid=0:0:40800>
0001114:INFO:cortex_m_v8m:CPU core #0 is Cortex-M33 r0p1 (security ext present)
0001130:INFO:cortex_m:FPU present: FPv5-SP
0001142:INFO:cortex_m_v8m:CPU core #1 is Cortex-M33 r0p1 (security ext present)
0001156:INFO:cortex_m:FPU present: FPv5-SP
0001167:INFO:dwt:4 hardware watchpoints
0001173:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001194:INFO:dwt:4 hardware watchpoints
0001200:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001226:INFO:server:Semihost server started on port 4444
0001226:INFO:gdbserver:GDB server started on port 50000
0001236:INFO:server:Semihost server started on port 4445
0001237:INFO:gdbserver:GDB server started on port 50001

Anyone has an idea what could be causing this behavior?

Thanks

haneefdm, 2019-05-06:
@danielRep, I know what the problem is... Cortex-Debug is looking for a regular expression that looks like the following [at line 129, pyocd.ts](https://github.com/Marus/cortex-debug/blob/ece08c0378fba77a74f13b864f2c011a7f5c7430/src/pyocd.ts#L129)
```
          /GDB server started at port/;
```

The output of your PyOCD is not outputting that. I see "GDB server started on port". Notice the "at" vs "on" so Cortex-Debug times out looking for the wrong output from PyOCD. The latest version of PyOCD uses "on" (my copy is the same as yours). Either wait for a fix or download source for Cortex-Debug and fix it yourself and install it. Or....

Not recommended, but you can edit the file ~/.vscode/extensions/marus25.cortex-debug-0.2.4/out/src/pyocd.js (line numbers will be different from the file I linked above) and see if that works.

I hope you only have to debug the first core? That is another issue I had, could not debug the second core.

danielRep, 2019-05-06:
Thanks @haneefdm ! That was indeed the problem. By editing the file in the directory specified from you I was able to establish a debug session.

For now I will only need to debug the first core, but I plan to use the two cores. Is there any special recommendation for that?

haneefdm, 2019-05-06:
@danielRep, glad it worked out. Far from ideal fix. I have many fixes like that, that I would like to merge but I am unable to. See issues #162 and #151. And yes, please don't close this issue until it is truly fixed. Also, it will be a reminder for me to fix this in my repo.

About the second processor, I have the same issue with both OpenOCD and PyOCD, we should look at it as a separate issue. I can give you a workaround but a real fix is not that simple. Once that issue is filed, then I can provide a workaround there. By that time, I hope to actually fix the problem with multiple processors

danielRep, 2019-05-06:
I will follow your fork, while @Marus is temporarily (I hope so) away.

flit, 2019-05-06:
Fwiw, both pyOCD and OpenOCD support a command argument that lets you echo a message of your choice. The launch config plugins for both tools in GNU MCU Eclipse use this to control the string that is being searched for, which reduces the dependency on built-in tool output. For both tools, the syntax is `-c 'echo blah blah'`. (I intentionally made pyOCD compatible with this argument from OpenOCD.)

Sorry that we broke the pyOCD support in cortex-debug. We were normalizing the server startup messages, and didn't realise it would break other tools.

haneefdm, 2019-05-07:
@flit Thanks for chiming in. One problem is that most gdbserver options are hardcoded in Cortex-Debug ... unlike Eclipse. I could be wrong, but there is no good way to add/edit the options sent to PyOCD (for OpenOCD, there is an option). I've done it before using a shell wrapper around PyOCD

@flit One question though. When do the -c commands get executed? And, if there are errors getting started, these commands do not get executed?

haneefdm, 2019-05-10:
Permanent fix coming soon... to adapt to the change in PyOCD

danielRep, 2019-05-13:
I will wait for that fix @haneefdm. Thanks for all the work and help!

Marus, 2019-05-15:
Just published V0.2.5 - should fix the PyOCD startup detection. If it doesn't work with this change please re-open the issue. And I apologize for the lack of speed in getting to this - haven't had much time to work on the project recently - but hopefully with a bit more free time and help from @haneefdm this will change in the future.

closed 2019-05-15 by Marus

danielRep, 2019-05-15:
It worked @Marus ! Thanks ! 👍