Marus/cortex-debug #167: FeatureRequest: Provide a gdbServerStarted regular expression entry in launch.json

haneefdm, 2019-05-15:
If we could have a user-override for knowing when the gdb-server has started properly and ready to take requests. Thing is until the gdb-server (OpenOCD/PyOCD/JLink/etc.) tells us it is okay to connect, we should not launch gdb because it will fail immediately or even worse intermittently.

This regex is really all Cortex-Debug looks for in server output and the servers seem to be changing a lot. The messages change if you use different(debug) options on certain debug-servers. Just last month or two, I saw several instances of failures in this area for a variety of reasons. Some have ancient gdb-servers or even custom modified ones.

Within the code for this extension, we call it `initMatch`. Going with a more descriptive name in launch.json is my preference. Fine with `initMatch` even.

There have been at least a half a dozen issues in recent days in both OpenOCD and PyOCD submitted with no easy workarounds

Having this feature could provide short term (or custom) workarounds. Please see #166, #165, #150 and many more

Marus, 2019-05-16:
Yeah - this seems like a good idea. We can have default ones provided that would cover our officially support versions (which we can try to keep up-to-date), but have the option to override. Will likely try to implement such an option this coming weekend - seems like a good option to provide flexibility.

haneefdm, 2019-05-16:
What do you think of looking for a port that is ready for use? If you had already tried it, could you share your experience? There are a couple of Node modules that could be useful.

Marus, 2019-05-16:
Well - it is supposed to do that already (finding the number of ports it needs starting at port 50000) - using a library called I believe portastic - unfortunately it has been less than reliable often returning ports to use that are already in use (I have seen it work as well, and having been able to determine what situations it works and doesn't work). Been meaning to look for alternative solutions to hopefully make that more stable

haneefdm, 2019-05-16:
I was thinking of something a bit different. 1) find a free port to use somehow 2) launch server asking it to use that port, 3) wait w/timeout to see that port started listening and accepting connections

Totally separate issue, but thought you might like to see:

I was playing with `portastic`. Near as I can tell, it doesn't do anything. Seems like an glorified counter :-) I started a debug session that consumed ports 50000 and 60000 by openocd, verified with. the Mac Network utility, `netstat` and `lsof` and asked portastic for available ports. I only used 50000 for debugging. Turns out 60001 is the other end (gdb) of the socket OS assigned. So, 50000 is listening and connected to 60001, 60000 is just listening with no one on the other end. Here is a glimpse.
```
HaneefMac-2018:cortex-debug.port hdm$ ./node_modules/portastic/bin/portastic find 50000 50010
Ports available to use: 50000, 50001, 50002, 50003, 50004, 50005, 50006, 50007, 50008, 50009, 50010

HaneefMac-2018:cortex-debug.port hdm$ netstat -a
Active Internet connections (including servers)
Proto Recv-Q Send-Q  Local Address          Foreign Address        (state)    
tcp4       0      0  localhost.50000        localhost.60001        ESTABLISHED
tcp4       0      0  localhost.60001        localhost.50000        ESTABLISHED
tcp4       0      0  localhost.50000        *.*                    LISTEN     
tcp4       0      0  localhost.60000        *.*                    LISTEN     

HaneefMac-2018:cortex-debug.port hdm$ sudo lsof -i :50000
Password:
COMMAND     PID USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME
openocd   78529  hdm    8u  IPv4 0x311de558d56e1655      0t0  TCP localhost:50000 (LISTEN)
openocd   78529  hdm    9u  IPv4 0x311de558d5869c55      0t0  TCP localhost:50000->localhost:60001 (ESTABLISHED)
arm-none- 78530  hdm    7u  IPv4 0x311de558d5867fd5      0t0  TCP localhost:60001->localhost:50000 (ESTABLISHED)
```
It didn't matter what numbers I asked `portastic`, it always says sure they are available. Not sure its definition of `available`. It doesn't do much. It relies on a module called bluebird. I asked for 0-100, so it printed 0-100 dutifully, same with pretty much any range I asked for. I know for sure quite a few of those busy.

I did not test Linux/Windows yet but this just told me why people submitted so many issues about ports.

noppej, 2019-05-17:
@haneefdm I can't help but wonder if the auto port assignment is more effort than it is worth. In general there are relatively few ports in use (between 0 and 6000), and all operating systems have an easy command to scan for available ports. My humble opinion is that it would be better to abandon auto port assignment, and just provide a launch.json option where a user can assign an available port for their use.   It will require 1 minute of extra effort during initial setup, but might save hours of trouble shooting later on. I am still using the same two ports I identified as available 2 years ago when I started with my embedded dev work. My $0.02 worth :)

haneefdm, 2019-05-17:
FYI: I found a replacement for `portastic`. Creating an API for drop-in replacement and testing on all platforms.

haneefdm, 2019-05-17:
@noppej 0-1023 are to be avoided because those have reserved uses. Many things will break. 22 for instance for ssh. It is recommended that we [use ports 49151 or above](https://en.wikipedia.org/wiki/Registered_port). And, here are also [ports to avoid](https://en.wikipedia.org/wiki/List_of_TCP_and_UDP_port_numbers)

So, yes it may be a good idea to have an override from launch.json but I will let @Marus decide. If the current node-module was working, then you should rarely see a problem. But alas. I think some gdb servers require up to three ports though. I have another situation where if we start openocd/pyocd on port 'X', the processor I want to debug is actually on port 'X+1' because that is the second processor in the chain.. See issue #152

haneefdm, 2019-05-17:
Port selection: I noticed is that some of these OS-affected modules are not actively maintained and are ancient. As new versions of OS show up, some just silently fail. I bet the module `portastic` which is just a wrapper around `bluebird` was working fine at one time. More than likely, it was a bug in the initial implementation but it seemed to work okay at the time. The chain of module dependencies can be huge and failures can happen anywhere.

I will try to create a unit test for my new implementation that will hopefully help identify issues early on.

Marus, 2019-05-18:
Yeah - I still like auto port selection (if we can get it working reliably), because as @haneefdm mentioned in some cases you can need up to three ports - which makes manual configuration more complex; can also make working in environments where you are working on multiple things at the same time easier as you then don't have to make sure your port configurations don't conflict. 

But yeah - while `portastic` did seem to work at some point - I think it is long since broken. Hopefully your alternative works out and provides are reliable alternative.

haneefdm, 2019-05-28:
I made a PR #169 to address the port selection issues of these issues

busticated, 2019-05-29:
is anyone actively working on the regex option for `launch.json`? if have some cycles i can devote to it if that'd be helpful.

cc @haneefdm @Marus

Marus, 2019-06-09:
The new `overrideGDBServerStartedRegex` options should have been released in the v0.2.7 release today. Had thought I was going to get this release out earlier - unfortunately some client issues ended up taking up too much of my time.

closed 2019-06-09 by Marus