Marus/cortex-debug #168: Double values displayed incorrectly

elagil, 2019-05-24:
Hello!

I am using cortex-debug with an STM32F746 and ST-Link V2 plus OpenOCD, built from source.

When using the variable view, I noticed incorrect display of double values. The value that is highlighted is actually 1.0 in the processor, and other calculations and another debug environment also show that. What can cause this?

[image]

Thanks in advance

haneefdm, 2019-05-28:
Would it be possible for you to provide the struct/typedef definition of what I am seeing the screenshot? I am assuming you are using the latest. I am also concerned about the -var-create: issue. I am assuming you are using the latest version V0.2.5

elagil, 2019-05-28:
I am using V0.2.6. Here comes the struct definition, but this also affects other double values in other places:

```
typedef struct
{
	bool en;
	uint32_t trig_cnt;
	uint32_t trig_cnt_tx;
	uint32_t trig_cnt_burst;
	uint32_t trig_cnt_subBurst;
	double rate;
	uint32_t triPeriod;
	uint32_t gptPeriod;
} trigger_set_t;
```

I tested with a JLink debugger and the problem persists.

The -var-create issue is harmless in my opinion, the variables are just not available in the context of the function that I stop the code in. They are in a different scope.

haneefdm, 2019-05-28:
So, this is not in the variables window but in the watch window. I missed that.

elagil, 2019-05-28:
Yes, I did not mention that precisely!

haneefdm, 2019-05-29:
I could not duplicate it with OpenOCD or JLink on my board. I am seeing the correct values.

In your launch.json, could you please enable debug output...
```
            "showDevDebugOutput": true,
```
and send the output from the debug console as well as the 'Adapter Output' section of the output tab. Clear the windows of any previous runs output though.

haneefdm, 2019-05-30:
@elagil was the value a result of a calculation right? I ask because value highlighted is pretty close to epsilon (a smallest double value that is > 0.0) which is `4.94065645841247E-324` for a double. So, that value can be considered close to `0`

You can take that value and use printf to find out the real value. Print using %e and %10.4f. I mean using your laptop/desktop program

So, what you are seeing may be the actual value, but some debuggers may be showing you rounded values?

haneefdm, 2019-05-30:
```C
#include <stdio.h>
#include <assert.h>
#include <float.h> 

double getEps (double m) {
  double approx=1.0;
  double lastApprox=0.0;
  while (m+approx!=m) {
    lastApprox=approx;
    approx/=2.0;
  }
  assert (lastApprox!=0);
  return lastApprox;
}
int main () {
  double val = 5.304989e-315;
  printf ("epsilon for 0.0 is %e\n", getEps (0.0));
  printf ("epsilon for 1.0 is %e\n", getEps (1.0));

  printf ("DBL_EPSILON is %e\n", DBL_EPSILON);

  printf ("Value is %e\n", val);
  printf ("Value is %10.8f\n", val);
  
  return 0;
}
```

haneefdm, 2019-05-30:
@elagil You can also find the address of rate field and examine the memory to make sure what the true value is

elagil, 2019-06-03:
You might be right, I tested with a minimal working example like this:

```
double v1 = 5.0;
uint32_t v2 = 3;
uint32_t v3 = v1 * v2;
```

and there seems to be a value of zero in memory (at least as seen by the debugger) for v1. I assumed caching to be the problem but I also tried in a region where caching was disabled in the memory protection unit (MPU) with the same result. Strangely, calculations are correct!

I will try the debugging output for GDB now.

[image]

elagil, 2019-06-03:
Find the adapter output and debug log attached.

[adapter_output.log](https://github.com/Marus/cortex-debug/files/3246401/adapter_output.log)
[debug.log](https://github.com/Marus/cortex-debug/files/3246402/debug.log)

haneefdm, 2019-06-03:
@elagil Thank you for reporting back. Dealing with floating point is always tricky **it floats :-)**, and when I program I always have the epsilon in mind for comparisons, but this is kind of extreme example you posted. I saw in the gdb output
```
GDB -> App: {"token":42,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["value","5.3114652946463512e-315"]]}}
```
I think think the GUI is displaying exactly what gdb reports. Not re-interpreting it.

I feel like the breakpt address is not in sync where we think the program stopped. Or it could be compiler optimizations or levels. Will sleep on it.

elagil, 2019-06-03:
Thanks for your comments. Build options are as follows, if it helps:
`-O0 -ggdb -fomit-frame-pointer -falign-functions=16 -lm`

Also, I use the ChibiOS real time operating system for STM32.

Marus, 2019-06-09:
This is certainly hard to determine what is going on here, as that is the value that GDB is returning to us.

I can't see anything particular odd in your build options that seem likely unfortunately, and have tried compiling programs with the same structure as you supplied here and double representation seems to come out correctly (both with and without those compile options); this was on a M4 arch as I don't have anything with an M7 core (doubt this is the cause - but I suppose it could be something with how GDB is interpreting things due to the hardware floating point support in the M7s; I've used doubles on M4Fs before as well, although not this particular structure).

What versions of GDB and the arm toolchain are you using?

elagil, 2019-10-11:
Sorry for the delay. I recently switched to a fresh install of ubuntu with the arm toolchain package 8-2019-q3 from the official pages. I need to check individual versions later.

The problem persists with ST link and jlink. I also verified it on a windows machine.

elagil, 2020-01-23:
The problem is definitely GDB related. When examining the memory location, the hex number can be converted to the correct double value with an external tool. How can it happen that GDB misinterprets the data?

E.g.:
`
p dds.dds_set->fref
$1 = 5.4493361955084617e-315
`

where fref should be 500e6. The memory dump shows:
`x/g &(dds.dds_set->fref)
0x20009d90 <dds_set>:	0x41bdcd6500000000`

which is indeed 500e6, when converted.

closed 2020-03-31 by haneefdm