Marus/cortex-debug #170: Feature Request: Register values display format not ideal?

haneefdm, 2019-05-28:
I suggest that register values should be shown closer to their natural format. Also, the setting of format should offer more choices -- perhaps mirroring basic C types (uint32, int32, float, double) in addition to 'Auto', 'Binary' and 'Hex'. Existing 'Decimal' could become 'int32'. I am not saying we should use these specific names.

In the registers window, everything is displayed in hex format by default. In general, this is good and if the user wants, they can change the format to Binary or Decimal

But the floating point registers should be displayed in their natural format. Also, there is no way to change the format to float or double. Very difficult to translate hex to a double/float

If I ask gdb to give me register values in the natural format 'N' as specified by the ABI, then I get most registers as decimal (not nice, but acceptable), but I do get single and double floats properly formatted. D0-D15 are doubles and S0-S31 refer to floats that alias into the same D0-D15

Another benefit I saw was when displaying the natural format, the PC will not only display the address but also has a label like `<main+16>` which can be helpful

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0439b/BEHDEEJE.html

What do you all think?

Marus, 2019-06-09:
This also sounds like a great idea - will look into it for the next update as well.

danielRep, 2019-07-04:
I find this useful, but I would like also to still have the option to change between formats. Please add that on the next release guys!

Marus, 2019-07-11:
Yes - looking at a newer option for formats that will play nicely with the updated system for the registers view. Hopefully will have that out there in the coming weeks (along with core register editing)

closed 2020-03-07 by haneefdm