Marus/cortex-debug #171: Register values don't track stack frames? Should they?

haneefdm, 2019-05-28:
I could be wrong, but the register values shown in CortexDebug are actually the values as they exist when the program is paused and they are in reference to the top of the stack -- the leaf.

When anything other than the top of the stack, register values do not seem to track like the Locals and Watch windows

GDB (and other debuggers). use the frame id to give you register values according to that context -- since the EABI calling convention says what registers are to be saved -- they can be found on the stack and gdb knows about these.

Please see:

https://sourceware.org/gdb/current/onlinedocs/gdb/Registers.html
https://sourceware.org/gdb/current/onlinedocs/gdb/Selection.html#Selection
https://sourceware.org/gdb/onlinedocs/gdb/GDB_002fMI-Stack-Manipulation.html

Marus, 2019-06-09:
This is a very good point - will work on this for the next release.

Marus, 2019-06-14:
This might be a bit harder than I had hoped; the locals and watch automatically make the update request when you change the stack frame. Still need to see if I can find a way to tie in and get events for selected stack frame changing so that could do the updates.

haneefdm, 2019-06-14:
I can take a shot at this later this afternoon or weekend.

haneefdm, 2019-06-14:
Every time the stack frame context changes, you get a `scopesRequest` I think. Which then triggers a `variablesRequest` I have to verify though.

Marus, 2019-06-15:
Yeah, that’s where I was thinking of tying in - not as ideal as a dedicated event for changing stack frames - but can probably pull something out of there and creat my own custom event triggered by one of those.

I wouldn’t start on this at the moment though, as I’m in the middle of a major refactoring of the code for both the core and peripheral register views, as the code was getting to be a bit of a tangled mess.

haneefdm, 2019-06-15:
Yup, not as good as a dedicated event. I was also halfway into it. Thought of a few ways of doing it. What I settled on was.

* using the scopesRequest, track the current scope -- which is threadId << 8 | frameId I think?
* created a 'customRequest in gdb.ts to return the current scope id
* in registers.ts, set a timer that starts on stoped and ends on continue or terminated events to polly every 100ms to see if the current scope changed. If so, do a refresh
* What I was working on was how to request gdb for registers for the particular context without doing a thread-select or frame-select. Docs say you can use --thread and --frame on all commands...

haneefdm, 2019-06-15:
@Marus I will wait for your refactoring, but wanted to see if this will work. Another, better method would be to for register.ts to listen on a socket so that whenever scopes change, we can notify on a socket. This would be nice and event-driven but need to create a socket.

haneefdm, 2019-06-15:
@Marus **Experiment failed** with both scopesRequest and variablesRequest. It appears that results are cached and it can only work once for a given scope after the debugger stops. No good way to update registers using this method. All I did was print a message when these requests are made.

Marus, 2019-06-15:
Yeah, I have a few thoughts of what might still work tying into some of those events- but want to finish this refactoring and cleanup (plus a few UX changes there) before I get into that part. Expect I’ll finish up my refactoring tonight or tomorrow - so will take a look at that then.

Marus, 2019-06-17:
Yeah- experimented with this - unfortunately don't think it will be possible to support currently - just don't have a way to tie into what we need (can get the first time they change to a frame, but if they go back to a frame they have already seen then some of the data is cached).

The underlying protocol used for the debug system in VSCode has recently added support for hardware registers - would place them into a special scope type in the main variables view - this isn't exposed in VSCode yet - but hopefully will be in the future - hopefully that would allow me to do this more properly (I had considered doing it as a scope before - but didn't think it worked well currently with the limitations of the variables view).

For now going to leave this - but hopefully will be something that can be further improved in the future.

haneefdm, 2019-06-17:
Hi @Marus, that was me that started and helped with the recent additions to the protocol :-) Also added support for Memory requests and disassembly to the protocol. The updated protocol was released a week or two ago. Took a couple of months -- mostly because of my delays and here was some length, heated debate.

I also prototyped registers in MIEngine and testing right now. This will test the protocol as well as create a reference implementation. A couple of MS Engineers looked at ways of implementing registers without disturbing the VSCode core and that is what we came up with. 

https://github.com/microsoft/debug-adapter-protocol/issues/41
https://github.com/microsoft/debug-adapter-protocol/issues/42

minkir014, 2019-07-21:
Now, both of them are closed.

PhilippHaefele, 2022-01-01:
This one should be resolved with new upcoming Release as the registers moved to variables view with PreRelease V1.1.2.

@haneefdm I will done some testing on Tuesday with the PreRelease. Anything preventing you from closing this issue already?

closed 2022-01-02 by haneefdm