Marus/cortex-debug #173: modify register values (and update gdb)

dagar, 2019-06-14:
One of the few remaining features keeping me (and other developers I know) on eclipse is the flexibility in interacting with the debugger when investigating a hardfault.

Basically I'm love to be able to recreate this workflow in vscode. https://www.element14.com/community/thread/54959/l/gdb-assisted-debugging-of-hard-faults?displayFullThread=true

Marus, 2019-06-14:
I’m actually currently working on things to be able to update register values (both core registers as well as memory mapped peripheral registers), hopefully this will be available in the next few weeks. Still trying to work a couple of things out in terms of how some of that will work.

dagar, 2019-06-14:
That's great to hear. I'm happy to test anything if that would help.

haneefdm, 2019-06-21:
@dagar, btw, all registers are available as $xxx so, until @Marus gets the UI implemented, you can use the gdb commands in the Debug Console. For instance, if you want to change r7, you can do

set $r7=4

The registers window will not refresh to show the new value but if you single step, you can and if does not modify r7, you will see the updated value for r7.

gdb prefixes all register values with a '$'

Hope this helps. You can also set memory contents this way if you know the address. SVD addresses for instance.

283958849, 2020-04-03:
Did this issue ( can not modify registers' value ) have implemented in released version 3.6 ?  In fact, it is very important feature. Thank you!

PhilippHaefele, 2021-12-25:
Link on element14 is down. https://mcuoneclipse.com/2015/07/05/debugging-arm-cortex-m-hard-faults-with-gdb-custom-command/ uses most likely a similar technic in it‘s GDB script/function. Still not sure if setting of registers is working in the meanwhile

PhilippHaefele, 2021-12-26:
> Did this issue ( can not modify registers' value ) have implemented in released version 3.6 ? In fact, it is very important feature. Thank you!

This is still not implemented. Most likely because of #171.

Moving registers to VARIABLES view ehivh will also allow setting the register values is on the [TODO list](https://github.com/Marus/cortex-debug/blob/master/TODO.md)

PhilippHaefele, 2022-01-01:
This one should be resolved with new upcoming Release as the registers moved to variables view with PreRelease V1.1.2.

@haneefdm I will done some testing on Tuesday with the PreRelease. Anything preventing you from closing this issue already

closed 2022-01-02 by haneefdm