Marus/cortex-debug #179: Unable to view Static/Global variables during debug

ryanplusplus, 2019-06-28:
Follow up to https://github.com/Marus/cortex-debug/pull/175#issuecomment-506534366

I built a tiny binary with a minimal main for two parts: an RX231 and an ATSAMD21G18A. This is `main.c` for both:

```c
int globalInt = 8;
static int staticInt = 9;

int main(void)
{
   while(1)
   {
      volatile int _volatile;
      _volatile = globalInt + staticInt;
      globalInt++;
      staticInt--;
   }

   return 0;
}
```

I debugged both binaries using a JLink (FINE interface for RX231, SWD for ATSAMD21G18A).

---

When debugging the RX231 part, the only thing to show up under Variables > Global was the name of a function argument in a stub function. Nothing showed up under Variables > Static. This is the relevant output from `objdump`:

```
rx-elf-objdump --syms out.elf | grep \\.data
00000004 l    d  .data  00000000 .data
00000004 l       .data  00000000 .LANCHOR0
00000008 l       .data  00000000 .LANCHOR1
00000008 l     O .data  00000004 _staticInt
00000004 l       .data  00000000 __clockSourceEnd
00000004 g       .data  00000000 __clockSourceStart
00000004 g     O .data  00000004 _globalInt
00000004 g       .data  00000000 _data
0000000c g       .data  00000000 _edata
```

I was able to verify with the memory viewer that `globalInt` and `staticInt` both contained their expected values at their expected offsets. I was also able to view `globalInt` and `staticInt` using a watch.

---

When debugging the ATSAMD21G18A part, `globalInt` was viewable under Variables > Global and showed the proper value. Nothing showed up under Variables > Static. This is the relevant output from `objdump`:

```
arm-none-eabi-objdump --syms out.elf | grep \\.data
20000000 l    d  .data  00000000 .data
20000000 l     O .data  00000004 ticks
2000000c l     O .data  00000004 staticInt
20000000 g       .data  00000000 __data_start__
20000008 g     O .data  00000004 globalInt
00000ac8 g       .data  00000000 __etext
20000004 g     O .data  00000004 SystemCoreClock
20000010 g       .data  00000000 __data_end__
```

I was able to view `globalInt` and `staticInt` using a watch.

---

It's not entirely surprising that globals aren't visible when debugging the RX231 part since it's not a Cortex M core, but there's nothing conspicuously different about the output of `objdump` except the names of the linker symbols.

It's unclear why no statics are visible in either case. This is the only part of the behavior that seems like it could be a real bug in the plugin.

ryanplusplus, 2019-06-28:
Full (non-grepped) `objdump` output for RX231:

```
out.elf:     file format elf32-rx-le

SYMBOL TABLE:
fff80000 l    d  .text  00000000 .text
fff81368 l    d  .rodata        00000000 .rodata
ffffff80 l    d  .fvectors      00000000 .fvectors
00000004 l    d  .data  00000000 .data
0000000c l    d  .bss   00000000 .bss
00000000 l    d  .debug_frame   00000000 .debug_frame
00000000 l    d  .debug_info    00000000 .debug_info
00000000 l    d  .debug_abbrev  00000000 .debug_abbrev
00000000 l    d  .debug_loc     00000000 .debug_loc
00000000 l    d  .debug_aranges 00000000 .debug_aranges
00000000 l    d  .debug_ranges  00000000 .debug_ranges
00000000 l    d  .debug_line    00000000 .debug_line
00000000 l    d  .debug_str     00000000 .debug_str
00000000 l    d  .comment       00000000 .comment
00000000 l    df *ABS*  00000000 Header_Pointers.c
00003665 l       .debug_abbrev  00000000 .Ldebug_abbrev0
00004154 l       .debug_line    00000000 .Ldebug_line0
00059f55 l       .debug_info    00000000 .Ldebug_info0
00000000 l    df *ABS*  00000000 build/avenger_rc17_pyramid/Source/Application/Hardware/RX2xx/start.o
fff80030 l       .text  00000000 L0
fff800b7 l       .text  00000000 _exit
fff800b9 l       .text  00000000 _loop_here
fff80030 l       .text  00000000 L0
fff800bb l       .text  00000000 L0
00000000 l    df *ABS*  00000000 r_dataflash.c
fff800bb l     F .text  00000022 _r_df_write_fpmcr
fff800bb l       .text  00000000 L0
fff800dd l       .text  00000000 L0
fff80102 l       .text  00000000 L0
fff8012c l       .text  00000000 L0
fff80138 l       .text  00000000 L0
fff801ab l       .text  00000000 L0
fff801f0 l       .text  00000000 L0
0000000c l       .bss   00000000 .LANCHOR0
0000000c l     O .bss   0000000c _data_flash_info
00000000 l       .debug_frame   00000000 .Lframe0
fff800bb l       .text  00000000 .LFB10
fff800dd l       .text  00000000 .LFB2
fff80102 l       .text  00000000 .LFB4
fff8012c l       .text  00000000 .LFB11
fff80138 l       .text  00000000 .LFB5
fff801ab l       .text  00000000 .LFB7
fff801f0 l       .text  00000000 .LFB9
00000000 l       .debug_abbrev  00000000 .Ldebug_abbrev0
00000000 l       .debug_ranges  00000000 .Ldebug_ranges0
00000000 l       .debug_line    00000000 .Ldebug_line0
00000000 l       .debug_loc     00000000 .LLST0
fff800e3 l       .text  00000000 .LVL4
fff800ee l       .text  00000000 .LVL5
00000022 l       .debug_loc     00000000 .LLST1
00000045 l       .debug_loc     00000000 .LLST2
00000066 l       .debug_loc     00000000 .LLST3
00000087 l       .debug_loc     00000000 .LLST4
000000a8 l       .debug_loc     00000000 .LLST5
000000d9 l       .debug_loc     00000000 .LLST6
000000fa l       .debug_loc     00000000 .LLST7
00000120 l       .debug_loc     00000000 .LLST8
00000143 l       .debug_loc     00000000 .LLST9
00000166 l       .debug_loc     00000000 .LLST10
fff8017a l       .text  00000000 .LVL19
fff801a5 l       .text  00000000 .LVL21
fff801ed l       .text  00000000 .LVL23
fff8022b l       .text  00000000 .LVL24
fff800bb l       .text  00000000 .LVL0
[... 921 lines cut ...]
fff80000 g     O .text  00000030 _sApplicationHeader
fff809f2  w    F .text  00000002 _TPU5_TCI5U
fff809f2  w    F .text  00000002 _Security_Error
fff809f2  w    F .text  00000002 _SCI1_TXI1
fff809f2  w    F .text  00000002 _TPU0_TGI0B
fff809f2  w    F .text  00000002 _SCI10_TEI10
fff809f2  w    F .text  00000002 _BSC_BUSERR
fff809f2  w    F .text  00000002 _CAC_OVFF
fff809f4  w    F .text  00000014 _PrivilegedInstructionException
fff80422 g     F .text  00000091 _S12AD_S12ADI0
00000040 g     O .bss   00000004 _flash_ready_isr_handler
fff809f2  w    F .text  00000002 _MTU1_TGIB1
fff80102 g     F .text  0000002a _R_DF_Write_Operation
fff80703 g     F .text  00000089 _SCI0_ERI0
fff809f2  w    F .text  00000002 _SCI12_TEI12
fff809f2  w    F .text  00000002 _USB0_USBI0
fff809f2  w    F .text  00000002 _SCI12_SCIX2
fff809f2  w    F .text  00000002 _TPU3_TGI3A
fff809f2  w    F .text  00000002 _SCI2_ERI2
fff80a9a g     F .text  0000001a _main
fff809f2  w    F .text  00000002 _CTSU_CTSURD
fff809f2  w    F .text  00000002 _SCI7_RXI7
fff809f2  w    F .text  00000002 _MTU2_TCIU2
fff809f2  w    F .text  00000002 _SCI3_ERI3
```

ryanplusplus, 2019-06-28:
Full (non-grepped) output of `objdump` for ATSAMD21G18A:

```
out.elf:     file format elf32-littlearm

SYMBOL TABLE:
00000000 l    d  .text  00000000 .text
20000000 l    d  .data  00000000 .data
20000010 l    d  .bss   00000000 .bss
00000000 l    d  .debug_info    00000000 .debug_info
00000000 l    d  .debug_abbrev  00000000 .debug_abbrev
00000000 l    d  .debug_aranges 00000000 .debug_aranges
00000000 l    d  .debug_ranges  00000000 .debug_ranges
00000000 l    d  .debug_line    00000000 .debug_line
00000000 l    d  .debug_str     00000000 .debug_str
00000000 l    d  .comment       00000000 .comment
00000000 l    d  .ARM.attributes        00000000 .ARM.attributes
00000000 l    d  .debug_frame   00000000 .debug_frame
00000000 l    d  .debug_loc     00000000 .debug_loc
00000000 l    df *ABS*  00000000 cortex_handlers.c
2000009c l     O .bss   00000004 usb_isr
00000000 l    df *ABS*  00000000 Reset.c
20000000 l     O .data  00000004 ticks
00000000 l    df *ABS*  00000000 WInterrupts.c
20000010 l     O .bss   00000044 ISRcallback
20000054 l     O .bss   00000044 ISRlist
20000098 l     O .bss   00000004 nints
00000000 l    df *ABS*  00000000 delay.c
200000a0 l     O .bss   00000004 _ulTickCount
00000000 l    df *ABS*  00000000 hooks.c
000001f8 l     F .text  00000004 __false
00000000 l    df *ABS*  00000000 startup.c
00000000 l    df *ABS*  00000000 wiring.c
00000000 l    df *ABS*  00000000 wiring_analog.c
00000000 l    df *ABS*  00000000 wiring_digital.c
00000000 l    df *ABS*  00000000 main.c
2000000c l     O .data  00000004 staticInt
00000000 l    df *ABS*  00000000 _thumb1_case_uqi.o
00000000 l    df *ABS*  00000000 _udivsi3.o
00000560 l       .text  00000000 .udivsi3_skip_div0_test
00000000 l    df *ABS*  00000000 _dvmd_tls.o
00000000 l    df *ABS*  00000000 variant.c
00000000 l    df *ABS*  00000000 
00000000 l       *UND*  00000000 countPulseASM
200000a4 l       .debug_info    00000000 __HeapLimit
20000000 g       .data  00000000 __data_start__
00000168  w    F .text  00000002 DMAC_Handler
00000168  w    F .text  00000002 HardFault_Handler
00000168  w    F .text  00000002 AC_Handler
000001d4 g     F .text  00000010 SysTick_Handler
20000008 g     O .data  00000004 globalInt
000001f8  w    F .text  00000004 sysTickHook
00000168  w    F .text  00000002 PendSV_Handler
00000168  w    F .text  00000002 NMI_Handler
00000ac8 g       .text  00000000 __exidx_end
00000168  w    F .text  00000002 TC7_Handler
00000ac8 g       .data  00000000 __etext
00000168  w    F .text  00000002 SERCOM1_Handler
00000168  w    F .text  00000002 ADC_Handler
00000000 g       .text  00000000 __text_start__
00000168  w    F .text  00000002 TCC1_Handler
00000168  w    F .text  00000002 SERCOM2_Handler
20000004 g     O .data  00000004 SystemCoreClock
00000168  w    F .text  00000002 TCC0_Handler
00000560 g     F .text  0000010a .hidden __udivsi3
00000168  w    F .text  00000002 RTC_Handler
20000010 g       .bss   00000000 __bss_start__
0000054c g     F .text  00000012 .hidden __gnu_thumb1_case_uqi
000000b4 g     F .text  00000078 tickReset
0000012c g     F .text  0000003c EIC_Handler
000004b4 g     F .text  00000078 pinMode
00000168  w    F .text  00000002 TC6_Handler
00000168  w    F .text  00000002 WDT_Handler
00000ac8 g       .text  00000000 __exidx_start
00000168  w    F .text  00000002 TC4_Handler
00000180 g     F .text  00000054 Reset_Handler
0000016c  w    F .text  00000014 USB_Handler
00000168  w    F .text  00000002 TC3_Handler
00000168 g     F .text  00000002 Dummy_Handler
000001e4 g     F .text  00000014 SysTick_DefaultHandler
20000010 g       .data  00000000 __data_end__
200000a4 g       .bss   00000000 __bss_end__
00000168  w    F .text  00000002 PM_Handler
00000168  w    F .text  00000002 SERCOM5_Handler
00000454 g     F .text  00000060 analogReference
0000036c g     F .text  000000e8 init
00000674  w    F .text  00000002 .hidden __aeabi_ldiv0
0000052c g     F .text  00000020 main
00000560 g     F .text  00000000 .hidden __aeabi_uidiv
00000168  w    F .text  00000002 SVC_Handler
00000168  w    F .text  00000002 TCC2_Handler
00000168  w    F .text  00000002 EVSYS_Handler
00000168  w    F .text  00000002 SERCOM3_Handler
200000a4 g       .debug_info    00000000 __end__
00000168  w    F .text  00000002 SERCOM4_Handler
00000168  w    F .text  00000002 I2S_Handler
00000168  w    F .text  00000002 NVMCTRL_Handler
000001fc g     F .text  00000170 SystemInit
00000168  w    F .text  00000002 SERCOM0_Handler
20008000 g       .bss   00000000 __StackTop
00000000 g     O .text  000000b4 exception_table
00000168  w    F .text  00000002 DAC_Handler
0000066c g     F .text  00000008 .hidden __aeabi_uidivmod
20007fff g       .bss   00000000 __ram_end__
00000678 g     O .text  00000450 g_APinDescription
20008000 g       *ABS*  00000000 __StackLimit
00000168  w    F .text  00000002 PTC_Handler
00000674  w    F .text  00000002 .hidden __aeabi_idiv0
00000000  w      *UND*  00000000 TC5_Handler
00000168  w    F .text  00000002 SYSCTRL_Handler
```

haneefdm, 2019-06-28:
@ryanplusplus Could you attach the full output as a file. The whitespace before the second hex value (size) is supposed to be a single tab or (white-space). What we see in the pasted stuff above is more than one space. That is what is causing a mismatch based on the code. If they are truly one or more spaces, the regex has to be adjusted. The man page does not specify.

You can try the following regexp in symbols.ts (replace the existing one)

`const SYMBOL_REGEX = /^([0-9a-f]{8})\s([lg\ !])([w\ ])([C\ ])([W\ ])([I\ ])([dD\ ])([FfO\ ])\s([^\s]+)\s+([0-9a-f]+)\s(.*)\r?$/;
`

ryanplusplus, 2019-06-28:
Here is the raw output from each:
[ATSAMD21G18A-objdump.txt](https://github.com/Marus/cortex-debug/files/3339620/ATSAMD21G18A-objdump.txt)
[RX231-objdump.txt](https://github.com/Marus/cortex-debug/files/3339626/RX231-objdump.txt)

I'll try the regex and report back.

ryanplusplus, 2019-06-28:
@haneefdm It looks like it is a single tab character before the size. As expected, the regex doesn't seem to change the behavior.

haneefdm, 2019-06-28:
@ryanplusplus yes, the original regex that Marus had worked on your file. Only thing I can think of is that the program did not actually execute. There is no error message if the program fails to run for any reason.

ryanplusplus, 2019-06-28:
@haneefdm if I wanted to see if I can figure out what is going wrong in the symbol parser, how can I see the output of, for example, a `console.log` while I'm testing the plugin?

haneefdm, 2019-06-28:
I am not sure how you are currently debugging the extension? You can use VSCode to debug the extension. Start by closing all other VSCode top level windows and just have the extension folder open. I am assuming you know how to compile and watch for changes in your source code.

In Debug panel, use the 'Extension + Debug Server' launch configuration. This will open another instance of VSCode with the Cortex-Debug extension.. Open your embedded program, edit launch.json, add the following line.

`            "debugServer": 4711,`

You will now be able to run debug both the backend and the extension frontend. They are running in separate programs (node instances) and they communicate via the port 4711. You can set breakpoints in either front end or backend or both and stop at appropriate points...

Temporarily, add the following code.. in symbol.ts right after the spawnSync call. and set a breakpoint at the spawnSync, single step -- I would have liked to put this code in there permanently except for the console.log

```javascript
            if (!objdump.output) {
                if (objdump.error) {
                    console.log(objdump.error);
                    throw objdump.error;
                }
                throw new Error(`Failed to get output from ${objdumpExePath}`);
            }
```
When you start this way, you now have two instances of VSCode windows. The first one contains the extension code and the second one contains you embedded project. Any console.logs()
will show up in the first VSCode windows 'Debug Console'. You will see output from either the front-end or the backend -- user drop down to select the 'Debug Server'

ryanplusplus, 2019-06-28:
Thanks @haneefdm. I was just using the "Launch Extension" debug configuration. I will set this up when I get some time to dig in.

Marus, 2019-06-28:
Yeah, as @haneefdm said - certainly seems to be an issue with it invoking the objdump executable properly for some reason, as the regex is properly matching against the output; so would have expected `staticInt` to show up in the statics scope and `globalInt` to show in the globals scope (with the _ prefix on the RX231)

If you can figure out why it may not be launching on your system we'd certainly be happy to get info and/or a PR to fix it.

Marus, 2019-06-29:
Oh - and as for the static not showing up on the SAMD part - were you actually stopped in the main paused in the main function at the time? Statics will only show up if your current scope is in the file that they are declared in.

Seems strange to have one work and not the other - as they both rely on the objdump functionality and the regex seems to be picking up.

ryanplusplus, 2019-06-29:
@marus yes, I was paused in `main.c` at the time. I agree, it struck me as odd as well. Hopefully it will be clear once I debug through the extension code.

haneefdm, 2019-06-29:
I would also put a conditional bkpt. like `name === 'staticInt'` to see what the rest of the matches look like.

[image]

haneefdm, 2019-06-30:
@ryanplusplus You may want to fetch/pull latest from this repo as there are changes for how launch.json is processed.

ryanplusplus, 2019-07-05:
@Marus, @haneefdm 

I was able to run this on the debugger and confirm that there are no errors running `objdump` and gettings its output.

These are the relevant entries in `this.symbols` for the RX target:

```
address:8
file:"main.c"
hidden:false
instructions:null
length:4
name:"_staticInt"
scope:0 // Local
section:".data"
type:2 // Object
```

```
address:4
file:null
hidden:false
instructions:null
length:4
name:"_globalInt"
scope:1 // Global
section:".data"
type:2 // Object
```

I followed things through and found that the issue in getting statics is that `this.symbolTable.getStaticVariables` is invoked with `"Source/Application/main.c"` but the symbol for `staticInt` indicates that the filename is just `"main.c"` so it is filtered out. I found the same results for statics on the Cortex M0 target (`"src/main.c"` vs. `"main.c"`).

The issue with globals appears to be related to GDB. It hits:

```javascript
if (err instanceof MIError && err.message === 'Variable object not found') {
```

in `gdb.ts` then throws again when it attempts to create the variable:

```
message:"-var-create: unable to create variable object"
name:"MIError"
```

I don't know nearly enough about MI to know how to go much further, but I assume that there's a dialect difference in terms of how variable names are generated for ARM and for RX.

ryanplusplus, 2019-07-05:
I did some googling and found that the issue with `-var-create` can be related to a disconnect between the version of `gcc` and the version of `gdb`. For now I'll assume that the maintainers of the GNU RX toolchain have done something strange.

haneefdm, 2019-07-05:
Just guesses on my part but, the file names in objdump/gdb depend on how the source files were compiled and linked. It also depends on what the current working dir. is for the debugger. Ideally, they should all be relative to your workspace/project directory or *yuk* absolute paths.

I don't know how gdb picked up `"Source/Application/main.c"`

With gdb, are you using any source maps or source paths? Some info about compiling from gcc man page.

-fdebug-prefix-map=old=new
When compiling files residing in directory old, record debugging information describing them as if the files resided in directory new instead. This can be used to replace a build-time path with an install-time path in the debug info. It can also be used to change an absolute path to a relative path by using . for new. This can give more reproducible builds, which are location independent, but may require an extra command to tell GDB where to find the source files. See also -ffile-prefix-map.

haneefdm, 2019-07-05:
It is almost like objdump and gdb are looking at very different executable files.

ryanplusplus, 2019-07-05:
I only have one `elf` in my tree so I don't think I'm looking at two different files.

`"Source/Application/main.c"` is the workspace relative path of the file and I confirmed that this is the path that is provided to `gcc`. We are not doing any prefix mapping.

When we build object files, they are built to a `build/` folder with the same relative path from the project root. That is, we build `"Source/Application/main.c"` to `"build/Source/Application/main.o"`. The linker is provided this path.

The `cwd` variable in my launch configuration is set to `${workspaceRoot}` and the executable is specified relative to `${workspaceRoot}`: `"${workspaceFolder}/build/<target name>/<target name>.elf"`.

As far as I can tell everything is workspace-relative so I'm not sure why `objdump` would provide anything but workspace-relative paths.

Interestingly, if I run `strings` on my binary I do see workspace-relative paths for every file except for `main.c`. However, even files that do get workspace-relative paths are given without any paths in `objdump`. I am, however, able to disassemble with source using `objdump` and it is able to find my source even though the paths are not given in the symbol output.

ryanplusplus, 2019-07-05:
This is the output from the debug console when trying to print `_globalInt`:

```
Breakpoint 
1, main () at Source/Application/main.c:11
11	      _volatile = globalInt + staticInt;
print _globalInt
No symbol "_globalInt" in current context. (from interpreter-exec console "print _globalInt")
No symbol "_globalInt" in current context.
print _staticInt
No symbol "_staticInt" in current context. (from interpreter-exec console "print _staticInt")
No symbol "_staticInt" in current context.
```

ryanplusplus, 2019-07-05:
If instead of using the mangled name I use the un-mangled name I get:

```
Breakpoint 
1, main () at Source/Application/main.c:11
11	      _volatile = globalInt + staticInt;
print globalInt
{"token":37,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
$1 = 8
```

ryanplusplus, 2019-07-05:
If I patch the code to also use the un-mangled symbol name then globals work for RX. So, for whatever reason, the GDB provided with the toolchain uses the un-mangled instead of mangled names.

It's probably not worth trying to work around this, but it would be nice to figure out why statics aren't working on ARM.

haneefdm, 2019-07-05:
It appears that objdump is showing you mangled names and gdb is showing you demangled names. For me, my object files themselves have demangled names for global variables.

Could you modify objdump. Normally, gdb has demangle set to auto. You can do a `show print asm-demangle` to show the setting.

You can force objdump to show -C to show demangled names but this should not be necessary if the obj. files themselves ae generated properly.

Guess: If you are using g++ to compile a .c file, you have to wrap your entire main file in an `extern "C" {}` or try just using gcc. Try the latter first.

ryanplusplus, 2019-07-05:
>It appears that objdump is showing you mangled names and gdb is showing you demangled names. For me, my object files themselves have demangled names for global variables.

I ran `objdump --syms` on `main.o` for RX and verified that the object file itself contains mangled names. They are demangled for the ARM build.

> You can force objdump ( -C ) to show demangled names but this should not be necessary if the obj. files themselves ae generated properly.

If I modify `cortex-debug` to pass `-C` to `objdump` then globals work correctly.

> Normally, gdb has demangle set to auto. You can do a show print asm-demangle to show the setting.

Here's the output of the demangle setting:

```
show print asm-demangle
{"token":27,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
Demangling of C++/ObjC names in disassembly listings is off.
```

> Guess: If you are using g++ to compile a .c file, you have to wrap your entire main file in an extern "C" {} or try just using gcc. Try the latter first.

It is compiling with `rx-elf-gcc` and if I try to add an `extern "C"` block it doesn't compile so it's definitely not compiling with `g++`.

---

My guess is that this fork of GCC does something non-standard with symbol names in its object files. I can work around this by providing a wrapper for `objdump` that forces the `-C` option.

haneefdm, 2019-07-05:
Not sure what is going on with this gcc. One thing, I do for globals is to put them in the Watch window because there are only a couple I am interested in at a time and I don't have to go hunt for them in the Variables window. That said, it would be nice if the globals and statics worked normally.

Yes, if you are indeed using gcc then it will complain about the extern "C", unless its guarded by the __cpplusplus ifdef.

```C++
#ifdef __cplusplus  
extern "C" { 
#endif 
    /* Declarations/code of this file */
#ifdef __cplusplus 
} 
#endif 
```

By the way, what I sent earlier should have been `show demangle-style`.
https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_57.html

ryanplusplus, 2019-07-05:
> By the way, what I sent earlier should have been `show demangle-style`.

I ran this and it shows `auto` as you expected:

```
show demangle-style
{"token":36,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
The current C++ demangling style is "auto".
```

ryanplusplus, 2019-07-05:
> Not sure what is going on with this gcc. One thing, I do for globals is to put them in the Watch window because there are only a couple I am interested in at a time and I don't have to go hunt for them in the Variables window. That said, it would be nice if the globals and statics worked normally.

Yeah this honestly isn't a big deal for my team because we can use watches and in general don't actually have any globals. I'm mostly interested in getting statics to work because I suspect that my team is not the only one with the path mismatch.

Frzzen, 2020-01-23:
same , unable to view Static variables with stm32f103 buid by arm-none-eabi-gcc.

Frzzen, 2020-01-23:
I push a PR to fix this #233 .
It solves my problem, and I hope it helps others

PhilippHaefele, 2022-01-05:
@haneefdm This one seems to be fixed from your side. Not 100% sure if code found it’s way into this repo, but i guess so. No feedback if it works as expected from @ryanplusplus & @frzzen yet

haneefdm, 2022-01-05:
Yes, this is fixed, was broken again and fixed again, improving performance a couple times (#539), etc. Wish we never had to rely on objdump.

We always demangle now and the option to control it is gone.

closed 2022-01-05 by haneefdm