Marus/cortex-debug #180: Separating launch/restart/attach override commands from SWO config.

haneefdm, 2019-07-05:
I was getting ready to use the new override commands for launch/restart/attach, but ran into an issue.

Currently, the built-in commands for these actions also do the SWO configuration. It could be a lot more useful if we separated the SWO configuration from the basic startup commands, or I have to kinda give up the SWO functionality (or replicate it) to use the overrides.

I can do the work for all gdb-servers, if this is acceptable. The overrides can then be a lot more useful. Opinions?

Marus, 2019-07-08:
This is a very good idea actually- didn’t think about that when I added the the override settings, but you are correct, as currently implemented they would likely break the SWO functionality. Would be happy to look at a PR for this.

haneefdm, 2019-07-08:
Thanks, @Marus. I will get to work on it tomorrow.

haneefdm, 2019-11-04:
This released in 0.3.3

closed 2019-11-04 by haneefdm