Marus/cortex-debug #181: Error: couldn't bind tcl to socket on port 6666: Address already in use...

busticated, 2019-07-09:
hey there :wave:

after rebuilding our `openocd` distributable to address https://github.com/Marus/cortex-debug/issues/166 i'm now noticing _new_ breakage.

it seems something is preventing `openocd` from closing its connection / releasing its socket. i'm able to launch the debugger using my custom `openocd` but upon stopping (via the "disconnect" button in VSCode's debug toolbar), i'm left with an open socket which can easily be identified via:

```
lsof -nP -i4TCP:6666 | grep LISTEN
```

which returns

```
_openocd 60812 me    3u  IPv4 0x5fdaf6926d9d699b      0t0  TCP 127.0.0.1:6666 (LISTEN)
```

attempting to start the debugger a second time results in:

```
Error: couldn't bind tcl to socket on port 6666: Address already in use
```

...in the "Adapter Output" panel

OS: macOS `v10.13.6`
VSCode: `v1.36.0`
cortex-debug: `v0.3.1`

haneefdm, 2019-07-09:
When you say 'disconnect' this the button used after an attach session? Did you notice if gdb was still running? But if gdb is still running, then OpenOCD will hang around. If that is the case, we have to figure out why and how this is happening. If you could tell me the steps to reproduce, I will surely look into it.

Yes, I have had situations where OpenOCD kept running even after terminating my debug session. But those were launch (not attach) sessions. It is very rare and I can't tell you what causes it. Maybe I unplugged the board without shutting down a debug session.

In those rare cases, I had to find and terminate OpenOCD process myself.

This extension does not deal with anything other than the gdb port. And even with the gdb port, it simply hands it off to gdb. The other two ports (telnet and tcl) are selected/used by OpenOCD itself. This extension or gdb do not use/mess with those ports

busticated, 2019-07-09:
thanks for taking a look @haneefdm :pray:

> When you say 'disconnect' this the button used after an attach session?

yes, the red square button in the group labeled "C" below

[image]

> Did you notice if gdb was still running? 

no, it's not. i can watch it launch and then exit when i start and stop my debugging session.

> This extension does not deal with anything other than the gdb port. And even with the gdb port, it simply hands it off to gdb. The other two ports (telnet and tcl) are selected/used by OpenOCD itself. This extension or gdb do not use/mess with those ports

ok. what do you do when the "stop" / "disconnect" button is clicked? just stop gdb? if you point me at some code maybe i can work things out 🤷‍♂

haneefdm, 2019-07-09:
See [code here](https://github.com/Marus/cortex-debug/blob/466e6bfec1ca6e3e52254bf9e1493c32edaf8b59/src/gdb.ts#L635)

We may have an issue here if it is an attach session, but yours is a launch session right?

busticated, 2019-07-09:
> See code here

thanks i’ll poke around in the morning

> yours is a launch session right?

no, it’s an `attach` session

haneefdm, 2019-07-09:
If it was a attach session, the extension will issue a detach command to gdb. This does not disconnect from the server (OpenOCD) or quit gdb. However, the extension waits for a second and (OS level) kills gdb -- and I am not sure gdb what kind of death gdb had -- don't it was able to clean up before its death.

This is what I meant earlier by there may be an issue with attach/detach. I was looking at that before I saw your message because I was debugging the attach/detach scenario for myself.

Another question. I am assuming you let the extension launch the OpenOCD server?

busticated, 2019-07-09:
@haneefdm 

>  I am assuming you let the extension launch the OpenOCD server?

yep! i just wire up my various custom paths, etc in `launch.json` and `settings.json` (btw, is it safe to migrate to `serverpath` as described in https://github.com/Marus/cortex-debug/issues/111#issuecomment-504778053?) and then press the big green triangle (labeled "B" in the image above)

busticated, 2019-07-09:
quick update: it seems the [code here](https://github.com/Marus/cortex-debug/blob/466e6bfec1ca6e3e52254bf9e1493c32edaf8b59/src/gdb.ts#L635) never runs for me though i'm plunking around in the minified bundles (`debugadapter.bundle.js` specifically) atm so 🤷‍♂

haneefdm, 2019-07-09:
Make sure you are up to date with the current HEAD of the source tree and compile everything. I had to clean everything since we moved to the new build system and start fresh or else, I was also ending up in the minified code.

In the Debug Console of VSCode where extension source is loaded, you can select which process you are debugging "Launch Extension" and "Debug Server". Make sure you select "Debug Server" (depending on which side you are debugging. I am assuming you used the "Extension + Debug Server" Launch configuration.

About your question for `serverpath` it works for me but not officially released in package.json yet.

busticated, 2019-07-10:
more info: the issue is _not_ happening over on windows (10, pro). i'm able to start and end a debugging session without leaving the openocd process hanging.

i suppose it's possible there's some bug in my `openocd` builds for darwin & linux.

busticated, 2019-07-10:
@haneefdm 

fssshhh... yeah... looks like my linux & darwin `openocd` builds were buggy 🤦‍♂🤦‍♂🤦‍♂😞 

with those knocked into shape, everything is working fine now. 

going to close this out. apologies for the noise. thanks for the help :pray:

closed 2019-07-10 by busticated