Marus/cortex-debug #182: GDB on windows is not able to match sources from debug information with WSL

fusedFET, 2019-07-09:
Suppose following scenario:
- Development PCs are on windows 10.
- Debugging is done with arm-none-eabi-gdb and a JLink on the windows side.
- The complete build toolchain is on a WSL or a build server.

The programmed ELF file contains all necessary debug informations.
However, the symbols refer to WSL paths: ("/mnt/c/…" instead of "C:/...").
Result is that Cortex-Debug is not able to locate the sources. 
In this state, the extension only allows stepping though the assembly code.

```
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"main () at /mnt/c/Users/[…]/src/main.c:7\n"}]}
main () at /mnt/c/Users/[…]/src/main.c:7
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"7\t/mnt/c/Users/[…]/src/main.c: No such file or directory.\n"}]}
```

I know the C/Cpp extension has a way to convert those symbols to "windows" paths, which allows debugging on x64. Is there a way to do this in Cortex debug?

haneefdm, 2019-07-09:
Cortex-Debug doesn't but gdb does as this is a very common problem. You can put commands in a .gdbinit file in your project/home dir or in your launch.json (many ways).

But, here is the [relevant documentation](https://sourceware.org/gdb/onlinedocs/gdb/Source-Path.html). If you want to use .gdbinit, [doc here](http://man7.org/linux/man-pages/man5/gdbinit.5.html). You can always source a file with all your custom commands.

[More info/tricks here](https://alex.dzyoba.com/blog/gdb-source-path/).

haneefdm, 2019-07-09:
Yes, cpptools does support a "sourceFileMap" which could be an enhancement request for this extension... but I think using gdb may be a better method. @Marus may want to comment on this

https://code.visualstudio.com/docs/languages/cpp#_locate-source-files

I am not sure why cpptools did this, perhaps because they support three different debuggers (gdb, lldb and clrdb)

fusedFET, 2019-07-09:
The workaround using gdb commands is perfect. Silly me, I should have checked out the doc beforehand.
Here a quick snippet of someone else has the problem for their launch.json:

```
            "postLaunchCommands": [
                "set substitute-path /mnt/c C:/"   
            ]
```

Marus, 2019-07-10:
I second @haneefdm 's recommendation that this be done through the pre/postLaunchCommands (or attach variants as appropriate). I don't really want to try and expose every possible GDB command through a dedicated configuration option - which is why we have the generic command ones (or always the possibility of a .gdbinit file).

I'm going to close this issue since I think you have an acceptable solution.

Marus, 2019-07-11:
Re-opened this issue as there may be a few other bits of functionality that benefit from mapping within the plugin (instead of strictly within the GDB launch commands) - but for now the work around (for most functionality) is the `set substitute-path` in `postLaunchCommands`/`postAttachCommands`

haneefdm, 2019-07-19:
There is a second problem I found with statics. objdump gives us an un-resolvable file-names that clearly gdb finds **without** any source maps. I have a case where none of the statics show up. I have a solution but need more time and will probably address the source map issue at the same time.

haneefdm, 2020-06-20:
Update: I fixed (I think) many issues with objdump, static variables/functions, etc.,and it also involved getting the wrong or no source files

Not saying the source mapping was all fixed. But I want to see specific examples that I can work on.

Closing the issue. Please re-open or start a new issue

closed 2020-06-20 by haneefdm

beyarkay, 2024-07-16:
> gdb

> The workaround using gdb commands is perfect. I should have checked out the doc before posting. Here a quick snippet if someone else has the problem for their launch.json:
> 
> ```
> "postLaunchCommands": [
>     "set substitute-path /mnt/c C:/"   
> ]
> ```

For future travellers: note that the windows paths have a *forward slash*, not a backslash. I think cortex-debug (or one of the other tools in this chain) automatically converts forward slashes to backwards slashes when it sees you're on windows, but it can't know what the correct full windows path is without the postLaunchCommand described above. So don't do what I did and try to be smart typing in ` "set substitute-path /mnt/c/ C:\"`, this won't work and will fail silently.