Marus/cortex-debug #183: Error: Unable to write to Folder Settings because cortex-debug.openocdPath does not support the folder resource scope...

busticated, 2019-07-09:
hi again :wave:

related to https://github.com/Marus/cortex-debug/issues/111 - i recently started hitting an error when attempting to set the `cortex-debug.openocdPath` configuration setting at the folder level within a multiroot workspace.

with a workspace layout like:

```
my-mutliroot-workspace
├── one/
│   ├── .vscode
│   │   └── settings.json
│   └── src
│       └── one.cpp 
├── two/
│   ├── .vscode
│   │   └── settings.json
│   └── src
│       └── two.cpp
├── three/
│   ├── .vscode
│   │   └── settings.json
│   └── src
│       └── three.cpp  
└── my-mutliroot-workspace.code-workspace
```

i'm trying to update the `cortex-debug.openocdPath` configuration setting for my users based on their _folder level_ `settings.json` data. when i try using:

```js
const config = workspace.getConfiguration('cortex-debug', projFolder.uri);
config.update('openocdPath', <path-goes-here>, ConfigurationTarget.WorkspaceFolder);
```

i get an error:

```
Error: Unable to write to Folder Settings because cortex-debug.openocdPath does not support the folder resource scope.
```

as it is, `cortex-debug` does not assign a `scope` value to the configuration options it contributes. as such, i believe a default of `"window"` is used ([see docs](https://code.visualstudio.com/api/references/contribution-points#configuration-scopes)). which means it can only be set at the "workspace" level. when working within a multiroot workspace, it'll be written to `my-mutliroot-workspace.code-workspace` and apply to all folders instead of each folder (e.g. `one/.vscode/settings.json`)

would it make sense for `"cortex-debug.openocdPath"` to use `"scope": "resource"` ? not sure but i would think some / all of the other exposed settings could use the same scope 🤷‍♂

haneefdm, 2019-07-09:
Haven't played with multi-root worskpaces but I thought folder level setings.json is not used (obsolete) in such a scenario

https://code.visualstudio.com/docs/getstarted/settings#_settings-file-locations

They become folder level settings ...

https://code.visualstudio.com/docs/editor/multi-root-workspaces#_settings

It is unclear to me what if any the extension has to do. I was hoping it would be transparent to the extension when it asks for its own configuration.

I am also thinking certainly, extensions should not be updating the users settings.json file?

busticated, 2019-07-09:
> I thought folder level setings.json is not used (obsolete) in such a scenario

no, they're used unless overridden at the workspace level (iow, set in your `.code-workspace` file). in some cases folder-level settings may conflict hence the need for `"scope"`. note that when launching a debugging session within a multiroot project you're given the option of selecting the _folder level_ `launch.json` configuration to use. similar UX for tasks - you're prompted to pick folder-specific tasks when working in a multiroot project. 

> I am also thinking certainly, extensions should not be updating the users settings.json file?

in this case, i agree which is why i filed https://github.com/Marus/cortex-debug/issues/111

busticated, 2019-07-09:
the easiest way to think about it is to imagine you are working on a product comprised of multiple hardware devices that work together. it's handy to be able to easily look at source code for each device. instead of opening multiple VSCode windows and loading each device's project folder into a single window and tabbing around, you can pull them all into a "multiroot workspace" - now you can open files from each project side-by-side and make edits to both within the same window.

now imagine one of your projects needs a different compiler version. if that can only be set at the workspace level, you'll need to actively manage it - changing it back and forth in order to build & debug your device firmware. build error in  project two? that's because `.code-workspace` is still using project two's settings. etc.

if instead `armToolchainPath` could be set at the project folder level, you'd simply set it there and only have to update it based on the needs of _that_ project.

haneefdm, 2019-07-09:
Thanks for the update. I should look at the VSCode APIs to understand better how it is all _supposed_ to work.

Personally, I am running into more people wanting to put `armToolchainPath` in launch.json (works by the way). It avoids the whole settings.json issue and OS variances for them.

That said, we need to figure out how an extension is supposed to behave/comply with multi-root workspaces and do the right thing. It may be that they do process the settings properly but not the stage where `CortexDebugConfigurationProvider` constructor (frontend) is called. We found out the hard way that they do not process the OS level overrides at that point but they do it by the time the Debug Server (backend) is invoked.

busticated, 2019-07-09:
> Personally, I am running into more people wanting to put armToolchainPath in launch.json (works by the way). It avoids the whole settings.json issue and OS variances for them.

yeah, i didn't mean to use that as an example (edited my post above) since as you say you can set it `launch.json` (and i am)

i was trying to illustrate _why_ the `scope` stuff matters generally since none of the [configuration contributions](https://github.com/Marus/cortex-debug/blob/master/package.json#L13-L70) use an explicit value.

> That said, we need to figure out how an extension is supposed to behave/comply with multi-root workspaces and do the right thing.

i _think_ you all are mostly ok as-is - exception being this `openocdPath` stuff - but i've only been using that code path so 🤷‍♂ 

naively, i would think for all but the obvious exceptions, you'd be fine setting `"scope": "resource"` across the board but again 🤷‍♂

Marus, 2019-07-11:
Hey @busticated - Can you try the .visx attached to this. This has the `scope` set to `resource` for most of the configuration settings that are there (there were a couple settings like enableTelemetry that seemed system wide) - just wanted to see if this actually fixes the issue for you. If it does we can include it in the next release - if not I'll need to do a bit more reading about the scope/multi-root stuff to see if I can come up with another solution.

[cortex-debug-0.3.2-pre1.vsix.zip](https://github.com/Marus/cortex-debug/files/3379823/cortex-debug-0.3.2-pre1.vsix.zip)

busticated, 2019-07-11:
@Marus thanks for poking at this :pray:

🤔 hm... debugging my extension and pausing within a command variable wired up to one of cortex-debug's settings, i'm getting odd results. i suspect something weird is happening w/ config resolution. i run the following in VSCode's debug console to grab data while paused:

```js
JSON.stringify(launchJSON, null, 4); // per-command variable behavior, `launchJSON` is the first arg to my command
JSON.stringify(vscode.workspace.getConfiguration('cortex-debug', projFolderUri), null, 4); // `projFolderUri` is a URI pointing to my project folder (e.g. `/Users/me/my-projects/multiroot/one`)
```

- - - -

**before**

<details>
<summary>launch JSON</summary>
<p>

```json
{
    "type": "cortex-debug",
    "request": "attach",
    "name": "Particle Debugger (photon, p1, electron)",
    "servertype": "openocd",
    "interface": "swd",
    "device": "STM32F205RG",
    "cwd": "/Users/me/my-projects/multiroot/one",
    "executable": "${command:particle.getDebuggerExecutable}",
    "preLaunchTask": "Particle: Flash application for debug (local)",
    "armToolchainPath": "${command:particle.getDebuggerCompilerDir}",
    "svdFile": "${command:particle.getDebuggerSVDFile}",
    "searchDir": [
        "${command:particle.getDebuggerSearchDir}"
    ],
    "configFiles": [
        "interface/cmsis-dap.cfg",
        "target/stm32f2x.cfg"
    ],
    "postAttachCommands": [
        "monitor reset halt"
    ],
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "runToMain": false,
    "serverpath": "${command:particle.getDebuggerOpenocdPath}",
    "toolchainPath": "${command:particle.getDebuggerCompilerDir}",
    "extensionPath": "/Users/me/Sites/particle/workbench/packages/marus25.cortex-debug-0.3.1"
}
```

</p>
</details>

<details>
<summary>settings JSON</summary>
<p>

```json
{
    "armToolchainPath": null,
    "JLinkGDBServerPath": null,
    "openocdPath": "${command:particle.getDebuggerOpenocdPath}",
    "pyocdPath": null,
    "PEGDBServerPath": null,
    "stutilPath": null,
    "enableTelemetry": true
}
```

</p>
</details>

- - - -
**after**

<details>
<summary>launch JSON</summary>
<p>

```json
{
    "type": "cortex-debug",
    "request": "attach",
    "name": "Particle Debugger (photon, p1, electron)",
    "servertype": "openocd",
    "interface": "swd",
    "device": "STM32F205RG",
    "cwd": "/Users/me/my-projects/multiroot/one",
    "executable": "${command:particle.getDebuggerExecutable}",
    "preLaunchTask": "Particle: Flash application for debug (local)",
    "armToolchainPath": "${command:particle.getDebuggerCompilerDir}",
    "svdFile": "${command:particle.getDebuggerSVDFile}",
    "searchDir": [
        "${command:particle.getDebuggerSearchDir}"
    ],
    "configFiles": [
        "interface/cmsis-dap.cfg",
        "target/stm32f2x.cfg"
    ],
    "postAttachCommands": [
        "monitor reset halt"
    ],
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "runToMain": false,
    "serverpath": null,
    "toolchainPath": "${command:particle.getDebuggerCompilerDir}",
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/Users/me/.vscode/extensions/marus25.cortex-debug-0.3.2-pre1"
}
```

</p>
</details>

<details>
<summary>settings JSON</summary>
<p>

```json
{
    "armToolchainPath": null,
    "armToolchainPrefix": "arm-none-eabi",
    "JLinkGDBServerPath": null,
    "openocdPath": "${command:particle.getDebuggerOpenocdPath}",
    "pyocdPath": null,
    "PEGDBServerPath": null,
    "stutilPath": null,
    "enableTelemetry": true
}
```

</p>
</details>

- - - -

note that the `serverpath` field is `null` in the launch JSON when running the updated `.vsix` but in `v0.3.1` it is set to the same value as `openocdPath` within my folder's `settings.json`. when running the `.vsix` if i _also_ set the `cortex-debug.openocdPath` field in my `.code-workspace` file, `serverpath` _is_ set properly in the launch JSON.

looking at the [code here](https://github.com/Marus/cortex-debug/blob/466e6bfec1ca6e3e52254bf9e1493c32edaf8b59/src/frontend/configprovider.ts#L171), i _think_ you can add safely the `folder` as the 2nd "resource" arg - e.g. `vscode.workspace.getConfiguration('cortex-debug', folder.uri);` ([docs](https://code.visualstudio.com/api/references/vscode-api#workspace.getConfiguration)) but i'm not sure.

of course, that seems like a somewhat significant change which probably warrants more testing - happy to help 👍

busticated, 2019-07-11:
taking a step back, maybe short-term the easy answer is "wait until the `serverpath` field is officially available within `launch.json`" since that really does offer the ideal UX for me here.

that said, i _do_ think it's worth shoring up scoping in general if only to avoid weird incompatibilities down the road.

either way, thanks very much for fielding these requests and for making an awesome and super-useful extension :pray::+1:

Marus, 2019-07-11:
Yeah - I'm thinking there will be an update maybe this weekend or early next week that will include exposing the `serverpath` setting in launch.json. I need to read up on the settings scopes - as that is quite substantially different from when I did the initial development of the extension. Will keep this open to track until I have a chance to look at that stuff in more detail.

busticated, 2019-07-11:
sounds good :+1:

fwiw, you are not alone. pretty much _every_ extension author had to make changes to accommodate this multiroot workspace stuff. having gone through it myself, i'd say it's a net win - they really do improve UX - but it's a bit of a pain.

haneefdm, 2021-12-22:
Don't think there is anything outstanding here, so closing it. `serverpath` is officially supported for a while now in lanch.json

closed 2021-12-22 by haneefdm