Marus/cortex-debug #185: Start Without Eebugging

luni64, 2019-07-13:
Just tried your extension with a Teensy 3.5 (MK64FX512). Everything works perfectly so far. Great work, will be VERY useful. 

The only thing I cant get working is "Start Without Debugging CTRL F5". Looks like it does exactly the same as "Start Debugging F5". Do I need to set up something or is it simply not implemented?

haneefdm, 2019-07-13:
Not sure what that means for an embedded program. If you disable all breakpoints, then start. I guess it is 'Start without debugging'? Also, Isn't it the same thing as hitting the reset button on your board?

Wonder what do other debuggers do when you do a 'Start without debugging' for embedded programs. For this extension, the only way to start a program is to use gdb and a server like OpenOCD or JLink.

luni64, 2019-07-13:
>  If you disable all breakpoints, then start. I guess it is 'Start without debugging'? 

Yes, this is what I want to achieve. However, 
1) it is not very convenient to disable the breakpoints first and 
2) It always pauses after F5: 
[image]
Only after another F5 the program starts.

> Also, Isn't it the same thing as hitting the reset button on your board?

Sure, however my board doesn't have a reset button, so I have to power cycle it...

Anyway, this is not a big deal at all, I'm just used to use Ctrl F5 to start and F5 to debug so I was wondering if I need to configure something.

haneefdm, 2019-07-13:
There was a big discussion on this in VSCode proper as well and not sure there were good answers/solutions on what is expected from a protocol point of view. As in which of the launch configuration parameters apply and what actions are allowable. For instance, breakpoints are handled/managed by VSCode and not the extension.

See: https://github.com/Microsoft/vscode/issues/2780

In a lot of cases, it was as simple as setting up a task in tasks.json to do the same without involving the debugger.

luni64, 2019-07-13:
Thanks for the link. That will do.

closed 2019-07-13 by luni64