Marus/cortex-debug #187: No disassembly view

thierryfischer, 2019-07-19:
Hello,

I cannot display any disassembly view of the currently executed code:
[image]
It would be great when the source code is not found (or debug informations not in elf) to display the assembler code as with the gdb command "disass $pc,+n".
_PS: Please note that I am using the Cortex Debug with a cpu different from arm with an external gdb server (Cortex Debug is the only extension I know which allows svd registers description)._

haneefdm, 2019-07-19:
Currently, disassembly is automatic for PC addresses that do not have any source file info, but it relies on having a known function name and it displays disassembly for the whole function.

Did you strip your executable to lose the debug symbols? I can understand not having source code for it.

thierryfischer, 2019-07-20:
There can be a lot of situations especially in embedded system where you do
not have debug information: can be a third party code, dynamically loaded
code, 'secret' library without debug, or just having forgot to compile with
-g or lost the elf.
My point is that in that case you do not need to know the current function
begining to display the next instructions.
Hence I was disapointed not having it has this is the case for Eclipse for
instance (actually Eclipse try to display what is before the pc too but it
is not 100% correct).

haneefdm, 2019-07-20:
Yes, I understand that you get libraries with no debug symbols. -- but they have some (all externs) or else you can't link to them.

About not having elf at all, can gdb even handle that? Cortex-Debug is 100% reliant on gdb, as in start a debug session. Suppose it can do it but I never tried it. I think we have to start there. If we cannot do it with gdb (and a gdb server), then this extension is not the right tool.

It is not that hard for this extension to do what eclipse does (print instrs relative to the PC), but your extreme use case of not having an elf stumped me. Priority wise, not sure where it stands.

haneefdm, 2019-07-20:
I have a plan to see about creating a better disassembly UX than what we have now and I will keep this in mind. But not thinking of the no-elf situation -- probably need at least a dummy one. Not happening soon though.

haneefdm, 2020-02-11:
This is way beyond what I can do, so I am closing it. The basic infrastructure in VSCode is lacking for this kind of debug. I am closing it, but if I ever figure out how to do it, I will. Trying to keep # of open issues down.

closed 2020-02-11 by haneefdm