Marus/cortex-debug #189: Breakpoints not working properly with WSL

FritzHaarmann, 2019-07-26:
Setup:

- Development PC is Windows 10
- Toolchain arm-none-eabi is on WSL
- Debugging is done using JLink on the Windows side

Behaviour:

- If I step through the code, I can add a breakpoint: it shows up in the BEAKPOINTS Tab with full absolute file path. The execution will stop at this breakpoint
- If I open a file in the Explorer View and add a breakpoint, it showa up in the BREAKPOINTS Tab but with relative path to the root folder. This breakpoint will be ignored

Expected behaviour:

- I can add breakpoints in the Explorer View and the  execution will stop at this breakpoint

haneefdm, 2019-07-26:
Couple of clarifications needed @FritzHaarmann. In the Debug Panel, you are seeing breakpoints with Windows pathnames or WSL pathnames? I am assuming Windows.

Have you adjust your pathnames (using source maps). Cortex Debug does not have an understanding of source maps yet (neither do I) and we did not know how gdb works when source maps exist. We discussed this recently in issue #182. For instance, from the command line (in Debug Console), does gdb accept a breakpoint in both Unix and Windows path names?

Oh WSL 1 or 2?

It would be nice to have WSL like scenarios (source maps really) supported but I am not sure if even MS has figured out what their long term strategy is. Please also see issue #66

FritzHaarmann, 2019-07-31:
Now I get the breakpoints work: I was just opening a directory in the first place, now I use a workspace and it seems to work. Now all breakpoints are with relative path.

 

Clarifications:

	I am using WSL 1
	I see breakpoints with WSL pathnames
	I don't use any source maps
	I cannot set breakpoints in gdb with Windows path names (C:\..., C:/...)

 

Gesendet: Freitag, 26. Juli 2019 um 14:43 Uhr

FritzHaarmann, 2019-07-31:
Oh, I forgot: The gdb I am using is also running on Linux (WSL). Only the JLink is on Windows since this is not supported by WSL.
 

Gesendet: Freitag, 26. Juli 2019 um 14:43 Uhr

haneefdm, 2020-06-20:
Please re-open and clarify the issue if this is still a problem. Too many things changed,

closed 2020-06-20 by haneefdm