Marus/cortex-debug #190: Failed to find open ports. Error: listen EADDRNOTAVAIL ::1:50000

BatuuKurt, 2019-07-31:
Setup: Windows 10 x64 / VSCode Insider 1.37.0

I want to build a docker cortex development environment. I connect to the container via the VSCode Extension [Remote Containers](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-containers).
The docker container has the arm-toolchain and JLink tools package installed.
The Cortex-Debug Extension is installed in Version 0.3.1

I use the following launch.json configuration:

    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug (J-Link)",
            "cwd": "${workspaceRoot}/Firmware",
            "executable": "./Build/TestProj.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "STM32F427VG",
            "interface": "swd",
            "ipAddress": "10.8.9.150"            
        }
    ]

The path to the JLink tools is in the PATH of the container.

On my host Windows machine I start JLink Remote Server which is waiting for client connections.
When I run the Debugger following error message appears: 

> Failed to find open ports. Error: listen EADDRNOTAVAIL ::1:50000

[image]

Otherwise I can run the following comand from the docker container shell:

    JLinkGDBServer -if swd -device STM32F427VG -select ip=10.8.9.150

It connects to the remote server as expected:

[image]

How can I get this running in VSCode with the Cortex-Debug Extension?

haneefdm, 2019-07-31:
I am not familiar with containers but it appears you want to run the JLink gdb-server on another host from where Cortex-Debug is running? If so, don't use `"servertype": "jlink"`. Try the following instead.
```javascript
            "servertype": "external",
            "gdbTarget": "<host>:2331",
```
Where `<host>` is either an IP address or a resolvable host-name. A concern I have is that Link probably creates a server on 127.0.0.1:2331 which may not be accessed by another computer (your container probably is a distinct host from your Windows host).

Please let us know if the above worked and did you have to do something more? There is also [this link](https://www.segger.com/products/debug-probes/j-link/tools/j-link-remote-server/) perhaps could be useful. Should not needed it if you can configure `JLinkGDBServer` to use actual IP address rather than the loopback.

Good luck.

BatuuKurt, 2019-07-31:
Hey @haneefdm ,
thank's for your response.
My intention is to run the JLinkGDBServer within the container, same host as the Cortex-Debug is running. For my understanding and following the description of your link, the remote Server on my Windows machine only provides the JLink probe via ethernet. A networked JLink adapter.

For my quick review of the code in the jlink.ts, the `serverExecutable()` returns the ` 'JLinkGDBServer'` string and the `serverArguments()` returns `cmdargs`  based on the `launch.json`. I think, it should build up the cli call `JLinkGDBServer -if swd -device STM32F427VG -select ip=10.8.9.150`.

Again, if I run this from the cli within the container, the remote connection will be established as expected.

I tried your advice with the `"servertype": "external"`. It does nothing ... no errors.

haneefdm, 2019-07-31:
The `external` means that you are going to manually launch a gdb-server on your local or any other machine and Cortex-Debug will not launch the server and istead just use what you specify in the `gdbTarget`. We need to check to see what host/port JLinkserver is actually using and use the same in `gdbTarget`

What is happening in your situation when you using `servertype=jlink` is that we are unable to find free ports to assign to JLink. Could be a permission problem. May I see that your network routing tables ... like `netstat -ir` to see what your loopback addresses look like.

I was wondering if your localhost address is being redirected in something like /etc/hosts. CortexDebug uses 0.0.0.0 to imply localhost. I want to send you a small node.js script to see what is going on but, I don't know what OS to code it for.

haneefdm, 2019-07-31:
Okay, it appears that docker does not support the default ipv6 loopback port '::1'. Sorry, I missed that earlier in the screenshot.

This was fixed a while ago on[ jun 25th](https://github.com/Marus/cortex-debug/commit/db20bd242efdec6f3f3ee88cf3d3f3a8680473fd#diff-f7e2bd2504fdeb3cf9c484ddc98b9cb4) but it is not in the latest cortex-debug release. Is it possible for you to use/build from the source?

BatuuKurt, 2019-08-01:
Docker for Windows do not support ipv6. Can you give me some instructions to build from source?

haneefdm, 2019-08-02:
Kinda hard to explain how to build and use. But,  if you wish, try this...I created a .vsix file for you from whatever is currently on this repo. Only way I could create this was to update the version to 0.3.2.

https://github.com/haneefdm/cortex-debug-samples/blob/master/cortex-debug-0.3.2.vsix

You can install manually using the command line
```
code --install-extension filename.vsix
```
Or in VSCode use `Extensions: Install from VSIX...` from the `Command Palette`

Not sure if you will get automated notices for updates or not. You can delete the extension and re-install later if need be. Check once in a while...

BatuuKurt, 2019-08-02:
Hey @haneefdm ,
thank you for this version. It seems to work. The remote server connection gets established as expected. Is your change going to make it into one of the upcomming releases?

haneefdm, 2019-08-02:
Yes, it is already on the master branch

closed 2019-08-02 by BatuuKurt