Marus/cortex-debug #193: MAX32620/1 SVD File

iwoloschin, 2019-08-07:
Has anyone gotten the MAX32620/1 SVD file to load?  Here's an example:

https://github.com/platformio/platform-maxim32/blob/develop/misc/svd/max32620.svd

I think I've got my config set properly, but maybe not?  Everything else appears to work fine, but trying to read some of the peripheral registers and that's not working.

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "executable": "build/max32620.elf",
            "name": "Debug (OpenOCD)",
            "device": "max32620",
            "configFiles": [
                "interface/ftdi/olimex-arm-usb-tiny-h.cfg",
                "target/max32620.cfg"
            ],
            "svdFile": "${workspaceRoot}/max32620.svd",
            "preLaunchTask": "Build Project"
        }
    ]
}
```

haneefdm, 2019-08-08:
May I ask what error message you are seeing if any? Although I am not using the same device, I was able to load your svd file and able to see the peripherals.

[image]

iwoloschin, 2019-08-08:
It says something like "No SVD File Found" which makes me think I might have incorrect syntax in my configuration?  I do not have access to the µC today so I'm not sure if I can test much, but I will have access tomorrow.

haneefdm, 2019-08-08:
No, that means pathname is not valid. Your launch.json syntax is fine. SVD file is fine. Maybe '${workspaceRoot}` is not where you think it is? Try the actual full path-name just for laughs?

iwoloschin, 2019-08-08:
I tried the fullpath yesterday as well without success, but I can try again tomorrow when I have access to the hardware (unless there's a way to run without hardware attached?).

Also, is there a good place to look for logs/errors/etc that might give more insight into what's happening?

haneefdm, 2019-08-08:
Unfortunately, not for this particular issue. You will see a little notification bubble pop up. but not sure it has lot of info.

I will see if I can a better message out to the debug console.

iwoloschin, 2019-08-09:
I have access to my hardware again and it seems that I am unable to load the SVD file with the full path or ${workspaceRoot}.  Happy to dig into this more if you can give me some pointers on where to look.

haneefdm, 2019-08-09:
I can't think of a way without you debugging the source. You will need a breakpoint here...
```javascript
                    reject(err);
```
[Link to source](https://github.com/Marus/cortex-debug/blob/466e6bfec1ca6e3e52254bf9e1493c32edaf8b59/src/frontend/svd.ts#L29) to start with

In my fork of the code, I have a better error message that percolates all the way to the Debug Console window, but not sure how to share that with you -- I am in the middle of other changes.

Silly thing but do you actually have permission to read that file? Perhaps it is owned by someone else...

iwoloschin, 2019-08-09:
Not a silly question!  I had the same thought and went and made triple sure I gave myself read/write access to the file!

If you have your own branch I can try to download that and install locally?

haneefdm, 2019-08-09:
Okay, give me a few mins and I will see if I can make a temp. release for you.

haneefdm, 2019-08-09:
You should now see a message of some sort as to why the svd could not be loaded. Download the following...

https://github.com/haneefdm/cortex-debug-samples/blob/master/cortex-debug-0.3.2.vsix

[See instructions here](https://github.com/haneefdm/cortex-debug-samples) and the warning.

iwoloschin, 2019-08-09:
Huh, weird.  It still says "No SVD File Loaded" with the path to the SVD file.  If I mistype the filename it'll pop up an error about ENOENT (which makes sense), but with the proper file name it's just not loading.

haneefdm, 2019-08-09:
Do you see anything in the 'Debug Console'?

Btw, you will see 'No SVD File Loaded' until the extension starts gdb successfully and is halted somewhere in your code. Were you at a breakpoint or paused?

iwoloschin, 2019-08-09:
Nothing in the debug console about svd files.  I'm sitting at a breakpoint now and still seeing the No SVD File Loaded message.  I won't have time to come back to this for a few days, but still interested in trying to debug more.

haneefdm, 2019-08-09:
Oh boy. Temporarily run out of ideas.

haneefdm, 2020-06-20:
@iwoloschin Can you confirm if this is still a problem with the latest release. There have been several changes made to the SVD infrastructure.

Please re-open and we can go further.

closed 2020-06-20 by haneefdm

iwoloschin, 2020-06-20:
Sorry, I'm no longer working on a project that would make use of this so I cannot confirm. Thanks for reaching out to check though!