Marus/cortex-debug #194: Write SWO to file or pipe

MabezDev, 2019-08-08:
A similar request to https://github.com/Marus/cortex-debug/issues/86 but I am using JlinkGDB server. The motivation being that vast amounts of data lags the console window heavily(and I've even had a vscode crash due to it), and also the lack of flexibility in displaying the data inside the console.

Is this currently possible, if not would it be possible to implement?

Awesome work by the way!

Marus, 2019-11-25:
This is something that I've started working on (as well as a few other tweaks to the SWO system). Will hopefully have a changes ready to go out in the next week or two.

PhilippHaefele, 2022-01-04:
@haneefdm As with #501 an implementation was done we can also close this one

haneefdm, 2022-01-04:
Yes, thank you.

closed 2022-01-04 by haneefdm