Marus/cortex-debug #195: How to pass parameters to JLinkGDBServer

bobatsar, 2019-08-12:
I would like to alter the RTTTelnet Port for my jlink to use it in parallell with another session.  [Search for RTTTelnetPort in Jlink Manual](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwj05b6p1fzjAhWNepoKHYHgABEQFjAAegQIABAC&url=http%3A%2F%2Fsupp.iar.com%2FFilesPublic%2FUPDINFO%2F011261%2Farm%2Fdoc%2Fjlinkarm.pdf&usg=AOvVaw0MH3KX_8-V5omp2Vmf-Sih)

I didn't find a way to pass additional parameters to the call of JLinkGDBServer so I tried to trick it with e.g.
`
"interface": "swd\" \"-RTTTelnetPort\" \"12345",
`
The command seems ok:

`
Launching server: "JLinkGDBServer" "-if" "swd" "-RTTTelnetPort" "12345" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "nrf52"
`
But the port is not open (checked with netstat). If I call the same command directly from the commandline, it works.

How can I pass parameters or is there another way to set the RTTTelnet port?

haneefdm, 2019-08-12:
Yes, the trick probably did not work. The whole thing sill went in as one string. Since the extension adds quotes and so did you, you can't tell.

Could you try using a shell script to manipulate the args and then launch (even exec) the gdb server?

haneefdm, 2020-02-11:
It should be in the next release. See #240

closed 2020-02-11 by haneefdm