Marus/cortex-debug #196: Run custom Jlink commands

5783mgordon, 2019-08-16:
I would like to run a custom jlink command to erase the flash before cortex-debug downloads new code.  Is that possible to add to the launch.json structure?

haneefdm, 2019-08-17:
One way I can think of is to use the `preLaunchTask` in launch.json to launch a task from tasks.json that will do the erase for you -- if JLink provides such a facility. Normally used to make sue executable is up to date but you can do both. If you try to do it from within the debug session, it will end the session rather abruptly.

danebou, 2020-02-06:
Yeah I need this functionality too. 

We Jlink you add the `erase` launch command before it starts up. 
```
            'interpreter-exec console "monitor halt"',
            'interpreter-exec console "monitor reset"',
            **'erase',**
            'target-download',
            'interpreter-exec console "monitor reset"',
            'enable-pretty-printing'
```

I'll submit a PR to see if they'll allow a `chipErase` option

haneefdm, 2020-02-06:
I think a lot has happened since my comment. Not everyone wants an erase because of target-download can be a smart download depending on the chip vendor and it is so much faster. Adding yet another option may not be wise either.

Please don't do a PR if existing functionality is sufficient. Have you tried `preLaunchCommands` where? You can do your own halt and erase. YMMV. If that is not sufficient...

There are several `launch` options now including using a JLink script. You can now control the entire launch sequence using  `overrideLaunchCommands` and `overrideRestartCommands`. Similar things exist for `attach`

I know the docs are a bit behind but VSCode IntelliSense capability is quite handy. That is how I found most of what I needed. Not the best way but, nobody reads docs anyway :-)

closed 2020-02-11 by haneefdm