Marus/cortex-debug #197: Feature Request: Ability to run a set of commands after a launch

haneefdm, 2019-08-18:
Soliciting feedback and I am willing to implement.

I would like to see a feature to add a set of commands that execute after a debugging session has started. This is different from `postLaunchCommands` as they are still run as part of the launch sequence. An example of that could be to issue a "continue". I have come across a situation where I have to do a
```
    "monitor gdb_sync",   // Tell OpenOCD to get ready to ignore the next stepi
    "stepi"               // Do a fake stepi and this forces gdb to sync with OpenOCD
```
to fix issues between OpenOCD and gdb. But if you do this as part of the launch sequence, breakpoints don't get set, GUI is in a confused state because it gets a 1-2 unexpected state transitions `running` and `stopped` because of the stepi. The `running` is not *real* so, anything Cortex-Debug does trusting that the program it is running will fail -- hard to recover.

There is a PR#186 that also needs such a feature - a feature that exists in Eclipse. But I am thinking, this could be a generic feature and the json way is better. We can have two json entries

```
"postStartCommands": [
    // This applies to both launch and attach
],
"postRestartCommands": [
    // This applies to both launch and attach
]
```
I realize, there are limits to what you can do in these commands and certain commands can kill the debug-session. But that is also true elsewhere.

Marus, 2019-09-04:
I'm liking this idea - good to provide a bit more flexibility.

haneefdm, 2019-09-06:
I propose new names `postStartSessionCommands` and `postRestartSessionCommands` so there is no conflict with existing names.

haneefdm, 2019-09-06:
Another change to the proposal. As I am implementing this, it is not making sense to have both `runToMain` enabled and this. `runToMain` takes precedence.

haneefdm, 2019-09-21:
Enhancement included in PR #206

haneefdm, 2019-10-15:
Released in V 0.3.3

closed 2019-10-15 by haneefdm

RogFedex, 2020-10-21:
With attach, after debugger is started the CPU is automatically paused in idle process and on continue (F5) CPU resumes. However if it is intended to automate resume process then expectation is to use the below launch configuration for the same.

"request": "attach",
"postStartSessionCommands" : ["continue"],

However, it is observed the Resume happens after debugger is connected but after automatic resume the debugger cannot be paused again.

In the debug console it is observed the continue command is noted, but the resume command has no effect
Continuing.
GDB -> App: {"token":17,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
18-stack-info-depth --thread 1 10000
19-stack-info-depth --thread 1 10000

FL0WL0W, 2021-06-21:
> With attach, after debugger is started the CPU is automatically paused in idle process and on continue (F5) CPU resumes. However if it is intended to automate resume process then expectation is to [...]

i found the solution is to use the command "-exec-continue" instead of "continue". Hope this helps anyone else looking for how to solve this problem.

Question for the project as a whole, I understand that the cpu must be halted at the start in order to sync up the debugger. But from an end user perspective, what is the purpose of halting at the beginning of the program? I have never had a use case for this and it has only been an annoyance to have to hit the continue button, If i needed to debug the program from the start i would just put a breakpoint there. Continuing should be the default, and i shouldn't have to dig down into the git issues to find a solutions. At least put this into the wiki.

haneefdm, 2021-06-21:
Doing a `continue` means that Cortex-Debug was unaware of the status of the debugger. VSCode thinks it is already paused so any further pauses will not be allowed by VSCode itself. Doing an `-exec-continue` is similar to Cortex-Debug doing it using GDB's machine interface and we receive a specially formatted packet back indicating success/failure. With `continue` we receive no such indication.

Default behavior after the initial halt is what is expected and recommended by gdb. I 

[The first thing GDB does after arranging to debug the specified process is to stop it. You can examine and modify an attached process with all the GDB commands that are ordinarily available when you start processes with run. You can insert breakpoints; you can step and continue; you can modify storage. If you would rather the process continue running, you may use the continue command after attaching GDB to the process.](https://sourceware.org/gdb/onlinedocs/gdb/Attach.html)

I have been using gdb like this for decades and normal thing to do is to examine what is going on (deadlocks, memories, peripherals, etc.) and perhaps set a global variable to enable logging or tracing and then let the program continue.

FL0WL0W, 2021-06-24:
The link you sent is for attaching to an ALREADY running process. That is not what we are doing here. we are starting from a reset. what am I going to examine? the reset values of all of the registers? why would you want to do this every time you run your project? This is not how eclipse or VisualGDB handles this and for good reason, it offers no benefit and only waists time.

haneefdm, 2021-06-24:
@FLOWLOW, there are many options for you. But first, the debugger has to be paused at the first instr. (whether you see it or not) to set breakpoints, RTOS detection, register detection, etc. This is done by the reset sequence

You can modify the startup sequence using `override...` in `launch.json`. What is there, by default is what most users want. That is not going to change w/o @Marus permission. There are people who debug startup/boot code, bootloaders, etc., as well. You can have exceptions even before you hit main. If you want to continue that is your choice and you can use the `-exec-continue` in the `post...` commands.

You can also use the `runToMain` or the upcoming `runToEntryPoint` to avoid (totally) stopping at the reset vector. Btw, this is also what Eclipse does for the vendor I use when you disable stopping in main. These are not auto-enabled by Cortex-Debug because some people have only two breakpoints and using one of these options burns one.